enioka-Haute-Couture/enioka_scan #95: Crash aléatoire lors du démarrage du scan par la caméra

abelatar, 2023-06-29:
Hello @marcanpilami ,
J'ai rencontré un problème où l'application se bloque parfois lors du démarrage du scan par la caméra. Le crash se produit avec l'exception suivante :
 ```
FATAL EXCEPTION: CameraBackground
                          Process: com.geodis.mobiprep, PID: 11551
                          java.lang.IllegalStateException: buffer is inaccessible
                          	at java.nio.DirectByteBuffer.get(DirectByteBuffer.java:231)
                          	at java.nio.ByteBuffer.get(ByteBuffer.java:557)
                          	at com.enioka.scanner.camera.CameraBarcodeScanViewV2.lambda$new$0$CameraBarcodeScanViewV2(CameraBarcodeScanViewV2.java:584)
                          	at com.enioka.scanner.camera.-$$Lambda$CameraBarcodeScanViewV2$VMyTEjvJxZ7LD_CakOd_WrWwxVc.onImageAvailable(Unknown Source:2)
                          	at android.media.ImageReader$ListenerHandler.handleMessage(ImageReader.java:798)
                          	at android.os.Handler.dispatchMessage(Handler.java:107)
                          	at android.os.Looper.loop(Looper.java:237)
                          	at android.os.HandlerThread.run(HandlerThread.java:67)
InputDispatcher        E  channel 'df5a119 com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity (server)' ~ Channel is unrecoverably broken and will be disposed!
WindowManager          E  win=Window{df5a119 u0 com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity EXITING} destroySurfaces: appStopped=false win.mWindowRemovalAllowed=true win.mRemoveOnExit=true win.mViewVisibility=0 caller=com.android.server.wm.AppWindowToken.destroySurfaces:1199 com.android.server.wm.AppWindowToken.destroySurfaces:1180 com.android.server.wm.WindowState.onExitAnimationDone:5030 com.android.server.wm.-$$Lambda$01bPtngJg5AqEoOWfW3rWfV7MH4.accept:2 java.util.ArrayList.forEach:1262 com.android.server.wm.AppWindowToken.onAnimationFinished:3539 com.android.server.wm.AppWindowToken.commitVisibility:859 
                       E  win=Window{200bc2b u0 Splash Screen com.geodis.mobiprep EXITING} destroySurfaces: appStopped=false win.mWindowRemovalAllowed=true win.mRemoveOnExit=true win.mViewVisibility=0 caller=com.android.server.wm.AppWindowToken.destroySurfaces:1199 com.android.server.wm.AppWindowToken.destroySurfaces:1180 com.android.server.wm.WindowState.onExitAnimationDone:5030 com.android.server.wm.WindowStateAnimator.onAnimationFinished:284 com.android.server.wm.WindowState.onAnimationFinished:5471 com.android.server.wm.-$$Lambda$yVRF8YoeNdTa8GR1wDStVsHu8xM.run:2 com.android.server.wm.SurfaceAnimator.lambda$getFinishedCallback$0$SurfaceAnimator:100 
InputDispatcher        E  Window handle Window{a402906 u0 com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity} has no registered input channel
WindowManager          E  RemoteException occurs on reporting focusChanged, w=Window{a402906 u0 com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity}
                          android.os.DeadObjectException
                          	at android.os.BinderProxy.transactNative(Native Method)
                          	at android.os.BinderProxy.transact(BinderProxy.java:527)
                          	at android.view.IWindow$Stub$Proxy.windowFocusChanged(IWindow.java:813)
                          	at com.android.server.wm.WindowState.reportFocusChangedSerialized(WindowState.java:3633)
                          	at com.android.server.wm.WindowManagerService$H.handleMessage(WindowManagerService.java:5203)
                          	at android.os.Handler.dispatchMessage(Handler.java:107)
                          	at android.os.Looper.loop(Looper.java:237)
                          	at android.os.HandlerThread.run(HandlerThread.java:67)
                          	at com.android.server.ServiceThread.run(ServiceThread.java:44)
BufferQueueProducer    E  [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@7b15eeb@0#0] cancelBuffer: BufferQueue has been abandoned
                       E  [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@7b15eeb@0#0] cancelBuffer: BufferQueue has been abandoned
                       E  [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@7b15eeb@0#0] cancelBuffer: BufferQueue has been abandoned
                       E  [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@7b15eeb@0#0] cancelBuffer: BufferQueue has been abandoned
                       E  [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@7b15eeb@0#0] cancelBuffer: BufferQueue has been abandoned
Finsky                 E  [9774] VerifyV31SignatureInstallTask.d(2): VerifyApps V31SignatureVerification: Successful verification for the package: com.geodis.mobiprep using APK Signature Scheme v3
SDAgentPa...eReceiver  E  Not going to handle 'com.geodis.mobiprep'!
                       E  Not going to handle 'com.geodis.mobiprep'!
WindowManager          E  win=Window{368c848 u0 Splash Screen com.geodis.mobiprep EXITING} destroySurfaces: appStopped=false win.mWindowRemovalAllowed=true win.mRemoveOnExit=true win.mViewVisibility=0 caller=com.android.server.wm.AppWindowToken.destroySurfaces:1199 com.android.server.wm.AppWindowToken.destroySurfaces:1180 com.android.server.wm.WindowState.onExitAnimationDone:5030 com.android.server.wm.WindowStateAnimator.onAnimationFinished:284 com.android.server.wm.WindowState.onAnimationFinished:5471 com.android.server.wm.-$$Lambda$yVRF8YoeNdTa8GR1wDStVsHu8xM.run:2 com.android.server.wm.SurfaceAnimator.lambda$getFinishedCallback$0$SurfaceAnimator:100 
InputDispatcher        E  Window handle Window{83fa13a u0 com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity} has no registered input channel
BufferQueueProducer    E  [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@cedd565@0#0] dequeueBuffer: BufferQueue has been abandoned
                       E  [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@cedd565@0#0] queueBuffer: BufferQueue(0x78b377de80) has been abandoned
Surface                E  queueBuffer: error queuing buffer to SurfaceTexture, -19 [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@cedd565@0#0]
BufferQueueProducer    E  [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@cedd565@0#0] queueBuffer: BufferQueue(0x78b377de80) has been abandoned
Surface                E  queueBuffer: error queuing buffer to SurfaceTexture, -19 [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@cedd565@0#0]
BufferQueueProducer    E  [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@cedd565@0#0] queueBuffer: BufferQueue(0x78b377de80) has been abandoned
Surface                E  queueBuffer: error queuing buffer to SurfaceTexture, -19 [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@cedd565@0#0]
WindowManager          E  RemoteException occurs on reporting focusChanged, w=Window{83fa13a u0 com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity}
                          android.os.DeadObjectException
                          	at android.os.BinderProxy.transactNative(Native Method)
                          	at android.os.BinderProxy.transact(BinderProxy.java:527)
                          	at android.view.IWindow$Stub$Proxy.windowFocusChanged(IWindow.java:813)
                          	at com.android.server.wm.WindowState.reportFocusChangedSerialized(WindowState.java:3633)
                          	at com.android.server.wm.WindowManagerService$H.handleMessage(WindowManagerService.java:5203)
                          	at android.os.Handler.dispatchMessage(Handler.java:107)
                          	at android.os.Looper.loop(Looper.java:237)
                          	at android.os.HandlerThread.run(HandlerThread.java:67)
                          	at com.android.server.ServiceThread.run(ServiceThread.java:44)
BufferQueueProducer    E  [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@cedd565@0#0] cancelBuffer: BufferQueue has been abandoned
                       E  [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@cedd565@0#0] cancelBuffer: BufferQueue has been abandoned
                       E  [SurfaceView - com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity@cedd565@0#0] cancelBuffer: BufferQueue has been abandoned
[... cut ...]
InputDispatcher        E  Window handle Window{8cbbc58 u0 Waiting For Debugger: com.geodis.mobiprep} has no registered input channel
                       E  Window handle Window{8cbbc58 u0 Waiting For Debugger: com.geodis.mobiprep} has no registered input channel
```

Étapes pour reproduire le problème (si possible) :

Ouvrir l'application Mobiprep.
Accéder à la fonctionnalité de scan par la caméra.
Observer que l'application peut se bloquer de manière aléatoire avec l'exception indiquée ci-dessus.

marcanpilami, 2023-06-29:
Hop ! Version beta3 ?

abelatar, 2023-06-30:
Oui enioka_scan-2.3.0-beta3.aar

marcanpilami, 2023-06-30:
Lié au matériel. La prochaine version va embarquer un contournement (catch retry). Par curiosité, sur quel device l'erreur se produit elle ?

closed 2023-06-30 by marcanpilami

abelatar, 2023-06-30:
sur un samsung XCover 4s

commit 34fb1bf2 (2023-06-30): Closes #95 - added workaround for lost camera2 buffer.