enioka-Haute-Couture/enioka_scan #100: Latence importante entre l'arrêt de la numérisation et le bip de lecture avec une exception

abelatar, 2023-07-05:
Hello @marcanpilami 

J'ai rencontré un problème lors de l'utilisation de la bibliothèque sur le TC27. Lorsque je fais des scans rapides, les bips se produisent immédiatement pour les premiers scans, puis il y a une grande latence avant que le bip de lecture avec succès réussie ne se produise.

J'utilise :
`scanner.getBeepSupport().beepScanSuccessful();`

Une exception est renvoyée dans les logs :

```

           E  changePosition 90200373002641256997 - 4
MobiprepActivity       E  beepScanSuccessful scanner
ScannerZebra           W  A scan was interrupted because (if null, means scanner has paused)
                          com.symbol.emdk.barcode.ScannerException: Already scanning. Wait for current scanning to complete.
                          	at com.symbol.emdk.barcode.Scanner.read(Scanner.java:526)
                          	at com.enioka.scanner.sdk.zebra.EmdkZebraScanner$AsyncStatusUpdate.doInBackground(EmdkZebraScanner.java:234)
                          	at com.enioka.scanner.sdk.zebra.EmdkZebraScanner$AsyncStatusUpdate.doInBackground(EmdkZebraScanner.java:215)
                          	at android.os.AsyncTask$3.call(AsyncTask.java:394)
                          	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
                          	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:305)
                          	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
                          	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
                          	at java.lang.Thread.run(Thread.java:1012)
AudioTrack             I  createTrack_l(115): AUDIO_OUTPUT_FLAG_FAST successful; frameCount 960 -> 960
ScannerService         D  Status changed: UNKNOWN --- Scanner is in an unknown status
                       D  Status changed: READY --- Scanner is ready to scan and waiting to be used
                       D  Status changed: SCANNING --- Scanner is in the process of scanning
Classific...nActivity  I  onData handled in ClassificationActivity
                       E  changePosition 90200173002640079995 - 4
MobiprepActivity       E  beepScanSuccessful scanner
ScannerZebra           W  A scan was interrupted because (if null, means scanner has paused)
                          com.symbol.emdk.barcode.ScannerException: Already scanning. Wait for current scanning to complete.
                          	at com.symbol.emdk.barcode.Scanner.read(Scanner.java:526)
                          	at com.enioka.scanner.sdk.zebra.EmdkZebraScanner$AsyncStatusUpdate.doInBackground(EmdkZebraScanner.java:234)
                          	at com.enioka.scanner.sdk.zebra.EmdkZebraScanner$AsyncStatusUpdate.doInBackground(EmdkZebraScanner.java:215)
                          	at android.os.AsyncTask$3.call(AsyncTask.java:394)
                          	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
                          	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:305)
                          	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
                          	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
                          	at java.lang.Thread.run(Thread.java:1012)
AudioTrack             I  createTrack_l(115): AUDIO_OUTPUT_FLAG_FAST successful; frameCount 960 -> 960
ScannerService         D  Status changed: UNKNOWN --- Scanner is in an unknown status
                       D  Status changed: READY --- Scanner is ready to scan and waiting to be used
                       D  Status changed: SCANNING --- Scanner is in the process of scanning
Classific...nActivity  I  onData handled in ClassificationActivity
                       E  changePosition 90200373002641173997 - 4
MobiprepActivity       E  beepScanSuccessful scanner
ScannerZebra           W  A scan was interrupted because (if null, means scanner has paused)
                          com.symbol.emdk.barcode.ScannerException: Already scanning. Wait for current scanning to complete.
                          	at com.symbol.emdk.barcode.Scanner.read(Scanner.java:526)
                          	at com.enioka.scanner.sdk.zebra.EmdkZebraScanner$AsyncStatusUpdate.doInBackground(EmdkZebraScanner.java:234)
                          	at com.enioka.scanner.sdk.zebra.EmdkZebraScanner$AsyncStatusUpdate.doInBackground(EmdkZebraScanner.java:215)
                          	at android.os.AsyncTask$3.call(AsyncTask.java:394)
                          	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
                          	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:305)
                          	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
                          	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
                          	at java.lang.Thread.run(Thread.java:1012)
AudioTrack             I  createTrack_l(115): AUDIO_OUTPUT_FLAG_FAST successful; frameCount 960 -> 960
ScannerService         D  Status changed: UNKNOWN --- Scanner is in an unknown status
                       D  Status changed: READY --- Scanner is ready to scan and waiting to be used
                       D  Status changed: SCANNING --- Scanner is in the process of scanning
Classific...nActivity  I  onData handled in ClassificationActivity
                       E  changePosition 90200173002654475998 - 4
MobiprepActivity       E  beepScanSuccessful scanner
ScannerZebra           W  A scan was interrupted because (if null, means scanner has paused)
                          com.symbol.emdk.barcode.ScannerException: Already scanning. Wait for current scanning to complete.
                          	at com.symbol.emdk.barcode.Scanner.read(Scanner.java:526)
[... 17 lines cut ...]
                          	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
                          	at java.lang.Thread.run(Thread.java:1012)
ToneGenerator          E  --- Stop timed out
AudioTrack             D  stop(150): called with 1152 frames delivered
libutils.threads       W  Thread (this=0xb4000070b3787e70): don't call waitForExit() from this Thread object's thread. It's a guaranteed deadlock!
ScannerZebra           W  A scan was interrupted because (if null, means scanner has paused)
                          com.symbol.emdk.barcode.ScannerException: Already scanning. Wait for current scanning to complete.
                          	at com.symbol.emdk.barcode.Scanner.read(Scanner.java:526)
                          	at com.enioka.scanner.sdk.zebra.EmdkZebraScanner$AsyncStatusUpdate.doInBackground(EmdkZebraScanner.java:234)
                          	at com.enioka.scanner.sdk.zebra.EmdkZebraScanner$AsyncStatusUpdate.doInBackground(EmdkZebraScanner.java:215)
                          	at android.os.AsyncTask$3.call(AsyncTask.java:394)
                          	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
                          	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:305)
                          	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
                          	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
                          	at java.lang.Thread.run(Thread.java:1012)
ToneGenerator          E  --- Stop timed out
AudioTrack             D  stop(151): called with 1152 frames delivered
libutils.threads       W  Thread (this=0xb4000070b377ef90): don't call waitForExit() from this Thread object's thread. It's a guaranteed deadlock!
ToneGenerator          E  --- Stop timed out
AudioTrack             D  stop(152): called with 192 frames delivered
libutils.threads       W  Thread (this=0xb4000070b3786910): don't call waitForExit() from this Thread object's thread. It's a guaranteed deadlock!
ToneGenerator          E  --- Stop timed out
AudioTrack             D  stop(153): called with 2880 frames delivered
```

marcanpilami, 2023-07-05:
Pourquoi apeller `scanner.getBeepSupport().beepScanSuccessful();` ? Une lecture avec succès va quoi qu'il arrive émettre un bip. Je regarde pour l'exception, mais elle est un peu tristement aussi présente dans les samples Zebra.

marcanpilami, 2023-07-05:
Si possible à restester avec la version 2.3.0

marcanpilami, 2023-07-06:
"scanner.getBeepSupport().beepScanSuccessful();  est utilisé pour le mode camera avant l'arrivé de mode scanner." => je ne comprends toujours pas. Un scan avec succès fait bip, caméra ou scanner physique. Pourquoi appeler cette API manuellement ?

La première trace est clairement pas avec la 2.3.0, il y a du avoir confusion car cette exception (par ailleurs sans conséquences) n'est plus lancée dans ce cas.

abelatar, 2023-07-07:
J'ai également désactivé les bips en cas d'erreur `beepScanFailure()` et cela a amélioré la fluidité de scans même en cas d'erreurs. Merci.

closed 2023-07-07 by abelatar