enioka-Haute-Couture/enioka_scan #101: Blocage de l'écran du TC27 et retard dans la lecture des codes-barres

abelatar, 2023-07-05:
Hello @marcanpilami 

Lors de l'utilisation de la bibliothèque sur le TC27, un problème survient qui bloque l'écran de l'appareil et entraîne un retard dans la lecture des codes-barres. Après avoir effectué plusieurs scans, l'écran du TC27 se fige, empêchant toute interaction avec l'application. De plus, le délai entre la tentative de lecture d'un code-barres et la réaction de l'application pour confirmer la lecture est anormalement long. Ce dysfonctionnement rend l'utilisation de la bibliothèque difficile et impacte négativement l'efficacité de la lecture des codes-barres sur le TC27.

Voici les traces : 

```
 I  Background young concurrent copying GC freed 270(152KB) AllocSpace objects, 0(0B) LOS objects, 0% free, 169MB/169MB, paused 891us,11.040ms total 100.641ms
BARCODE                D  FPS: 23.377989 - Pool size: 6. Current analysis queue depth: 2. Current res: 3264*2448. Statistics are: {Point(3264, 2448)=65}
geodis.mobiprep        I  Background concurrent copying GC freed 544(168KB) AllocSpace objects, 22(100MB) LOS objects, 17% free, 112MB/136MB, paused 99us,106us total 205.461ms
                       I  Background concurrent copying GC freed 464(155KB) AllocSpace objects, 18(76MB) LOS objects, 20% free, 95MB/119MB, paused 221us,92us total 147.994ms
                       I  Background young concurrent copying GC freed 238(153KB) AllocSpace objects, 0(0B) LOS objects, 0% free, 146MB/146MB, paused 563us,4.005ms total 143.016ms
                       I  Background concurrent copying GC freed 529(164KB) AllocSpace objects, 21(79MB) LOS objects, 20% free, 92MB/116MB, paused 172us,49us total 149.600ms
BARCODE                D  FPS: 21.65107 - Pool size: 6. Current analysis queue depth: 3. Current res: 3264*2448. Statistics are: {Point(3264, 2448)=66}
geodis.mobiprep        I  Background young concurrent copying GC freed 293(137KB) AllocSpace objects, 0(0B) LOS objects, 0% free, 151MB/151MB, paused 6.159ms,4.138ms total 197.656ms
                       I  Background concurrent copying GC freed 611(166KB) AllocSpace objects, 16(81MB) LOS objects, 17% free, 111MB/135MB, paused 137us,194us total 212.325ms
ScannerService         D  Status changed: SCANNING --- Scanner is in the process of scanning
ScannerActivity        D  Received barcode from scanner: 90200173002640079995 - CODE128
ParcelLoadingActivity  I  onData handled in ParcelLoadingActivity
AudioTrack             I  createTrack_l(161): AUDIO_OUTPUT_FLAG_FAST successful; frameCount 960 -> 960
geodis.mobiprep        I  Background young concurrent copying GC freed 386(199KB) AllocSpace objects, 0(0B) LOS objects, 0% free, 144MB/144MB, paused 175us,165us total 102.892ms
ImageReader_JNI        W  Unable to acquire a buffer item, very likely client tried to acquire more than maxImages buffers
                       W  Unable to acquire a buffer item, very likely client tried to acquire more than maxImages buffers
ScannerZebra           W  A scan was interrupted because (if null, means scanner has paused)
                          com.symbol.emdk.barcode.ScannerException: Already scanning. Wait for current scanning to complete.
                          	at com.symbol.emdk.barcode.Scanner.read(Scanner.java:526)
                          	at com.enioka.scanner.sdk.zebra.EmdkZebraScanner$AsyncStatusUpdate.doInBackground(EmdkZebraScanner.java:234)
                          	at com.enioka.scanner.sdk.zebra.EmdkZebraScanner$AsyncStatusUpdate.doInBackground(EmdkZebraScanner.java:215)
                          	at android.os.AsyncTask$3.call(AsyncTask.java:394)
                          	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
                          	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:305)
                          	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
                          	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
                          	at java.lang.Thread.run(Thread.java:1012)
ParcelLoadingActivity  E  Problem camera 306 : Attempt to invoke virtual method 'void com.enioka.scanner.sdk.camera.CameraBarcodeScanViewScanner.resume()' on a null object reference
ScannerService         D  Status changed: UNKNOWN --- Scanner is in an unknown status
                       D  Status changed: READY --- Scanner is ready to scan and waiting to be used
ImageReader_JNI        W  Unable to acquire a buffer item, very likely client tried to acquire more than maxImages buffers
BARCODE                W  Discarding image as all analysers are busy
                          java.lang.IllegalStateException: maxImages (12) has already been acquired, call #close before acquiring more.
                          	at android.media.ImageReader.acquireNextImage(ImageReader.java:636)
                          	at android.media.ImageReader.acquireLatestImage(ImageReader.java:507)
                          	at com.enioka.scanner.camera.CameraBarcodeScanViewV2.lambda$new$0$CameraBarcodeScanViewV2(CameraBarcodeScanViewV2.java:609)
                          	at com.enioka.scanner.camera.-$$Lambda$CameraBarcodeScanViewV2$VMyTEjvJxZ7LD_CakOd_WrWwxVc.onImageAvailable(Unknown Source:2)
                          	at android.media.ImageReader$1.run(ImageReader.java:916)
                          	at android.os.Handler.handleCallback(Handler.java:942)
                          	at android.os.Handler.dispatchMessage(Handler.java:99)
                          	at android.os.Looper.loopOnce(Looper.java:201)
                          	at android.os.Looper.loop(Looper.java:288)
                          	at android.os.HandlerThread.run(HandlerThread.java:67)
ImageReader_JNI        W  Unable to acquire a buffer item, very likely client tried to acquire more than maxImages buffers
BARCODE                W  Discarding image as all analysers are busy
                          java.lang.IllegalStateException: maxImages (12) has already been acquired, call #close before acquiring more.
                          	at android.media.ImageReader.acquireNextImage(ImageReader.java:636)
                          	at android.media.ImageReader.acquireLatestImage(ImageReader.java:507)
                          	at com.enioka.scanner.camera.CameraBarcodeScanViewV2.lambda$new$0$CameraBarcodeScanViewV2(CameraBarcodeScanViewV2.java:609)
                          	at com.enioka.scanner.camera.-$$Lambda$CameraBarcodeScanViewV2$VMyTEjvJxZ7LD_CakOd_WrWwxVc.onImageAvailable(Unknown Source:2)
                          	at android.media.ImageReader$1.run(ImageReader.java:916)
                          	at android.os.Handler.handleCallback(Handler.java:942)
                          	at android.os.Handler.dispatchMessage(Handler.java:99)
                          	at android.os.Looper.loopOnce(Looper.java:201)
                          	at android.os.Looper.loop(Looper.java:288)
                          	at android.os.HandlerThread.run(HandlerThread.java:67)
AudioTrack             I  createTrack_l(115): AUDIO_OUTPUT_FLAG_FAST successful; frameCount 960 -> 960
ImageReader_JNI        W  Unable to acquire a buffer item, very likely client tried to acquire more than maxImages buffers
                       W  Unable to acquire a buffer item, very likely client tried to acquire more than maxImages buffers
BARCODE                W  Discarding image as all analysers are busy
                          java.lang.IllegalStateException: maxImages (12) has already been acquired, call #close before acquiring more.
                          	at android.media.ImageReader.acquireNextImage(ImageReader.java:636)
                          	at android.media.ImageReader.acquireLatestImage(ImageReader.java:507)
[... 54 lines cut ...]
                          	at android.os.HandlerThread.run(HandlerThread.java:67)
geodis.mobiprep        I  Background concurrent copying GC freed 1620(278KB) AllocSpace objects, 17(55MB) LOS objects, 13% free, 147MB/171MB, paused 13.910ms,2.328ms total 584.164ms
ImageReader_JNI        W  Unable to acquire a buffer item, very likely client tried to acquire more than maxImages buffers
BARCODE                D  FPS: 12.606225 - Pool size: 6. Current analysis queue depth: 5. Current res: 3264*2448. Statistics are: {Point(3264, 2448)=67}
                       I  Changing preview resolution from 3264*2448 to 2560*1920
                       D  New preview resolution set
[... cut ...]
                          	at android.os.Handler.dispatchMessage(Handler.java:99)
                          	at android.os.Looper.loopOnce(Looper.java:201)
                          	at android.os.Looper.loop(Looper.java:288)
                          	at android.os.HandlerThread.run(HandlerThread.java:67)
ImageReader_JNI        W  Unable to acquire a buffer item, very likely client tried to acquire more than maxImages buffers
                       W  Unable to acquire a buffer item, very likely client tried to acquire more than maxImages buffers
```

marcanpilami, 2023-07-05:
Si possible à retester avec la version 2.3.0

abelatar, 2023-07-06:
Bonjour Marc-Antoine, 
Oui cela existe toujours même avec la version 2.3.0 , je viens de tester sur un Xcover 4S et il y a le même problème, de plus en mode portrait c'est quasi impossible de scanner sur un Xcover (la prévisu de la camera semple compressé horizontalement)
 
Voici les traces sur Xcover : 
```

createTrack_l(0): AUDIO_OUTPUT_FLAG_FAST successful; frameCount 960 -> 960
                       D  setVolume(1.000000, 1.000000) pid : 9234
                       D  stop(25): called with 1920 frames delivered
BARCODE                D  FPS: 19.802664 - Pool size: 8. Current analysis queue depth: 0. Current res: 3840*2160. Statistics are: {Point(3840, 2160)=1}
                       D  barcode read: 90200371338350675999
CamBarcod...nVScanner  V  handleScanResult 90200371338350675999 - CODE128
Classific...nActivity  I  onData handled in ClassificationActivity
                       D  first scan on batch mode
geodis.mobipre         I  Background concurrent copying GC freed 735(196KB) AllocSpace objects, 13(59MB) LOS objects, 15% free, 129MB/153MB, paused 4.905ms total 444.521ms
                       I  WaitForGcToComplete blocked Alloc on HeapTrim for 342.573ms
                       I  Starting a blocking GC Alloc
AudioTrack             I  createTrack_l(-1005584384): AUDIO_OUTPUT_FLAG_FAST successful; frameCount 960 -> 960
                       D  setVolume(1.000000, 1.000000) pid : 9234
                       D  stop(26): called with 1024 frames delivered
geodis.mobipre         I  Waiting for a blocking GC Alloc
                       I  Background concurrent copying GC freed 1011(258KB) AllocSpace objects, 15(68MB) LOS objects, 16% free, 121MB/145MB, paused 205us total 205.411ms
                       I  WaitForGcToComplete blocked Alloc on HeapTrim for 63.477ms
                       I  Starting a blocking GC Alloc
                       I  Waiting for a blocking GC Alloc
                       I  Background concurrent copying GC freed 804(178KB) AllocSpace objects, 21(64MB) LOS objects, 16% free, 124MB/148MB, paused 1.535ms total 276.926ms
                       I  WaitForGcToComplete blocked Alloc on HeapTrim for 71.910ms
                       I  Starting a blocking GC Alloc
                       I  Waiting for a blocking GC Alloc
                       I  Background concurrent copying GC freed 981(181KB) AllocSpace objects, 12(66MB) LOS objects, 16% free, 117MB/141MB, paused 858us total 181.582ms
                       I  WaitForGcToComplete blocked Alloc on HeapTrim for 34.025ms
                       I  Starting a blocking GC Alloc
BARCODE                D  FPS: 20.238422 - Pool size: 8. Current analysis queue depth: 4. Current res: 3840*2160. Statistics are: {Point(3840, 2160)=2}
geodis.mobipre         I  Waiting for a blocking GC Alloc
                       I  Background concurrent copying GC freed 685(193KB) AllocSpace objects, 13(52MB) LOS objects, 15% free, 133MB/157MB, paused 216us total 224.306ms
                       I  WaitForGcToComplete blocked Alloc on HeapTrim for 15.964ms
                       I  Starting a blocking GC Alloc
                       I  Waiting for a blocking GC Alloc
                       I  Background concurrent copying GC freed 728(176KB) AllocSpace objects, 10(50MB) LOS objects, 14% free, 136MB/160MB, paused 9.331ms total 206.732ms
                       I  WaitForGcToComplete blocked Alloc on HeapTrim for 61.994ms
                       I  Starting a blocking GC Alloc
                       I  Waiting for a blocking GC Alloc
                       I  Background concurrent copying GC freed 658(191KB) AllocSpace objects, 12(51MB) LOS objects, 14% free, 137MB/161MB, paused 3.202ms total 306.523ms
                       I  WaitForGcToComplete blocked Alloc on HeapTrim for 185.525ms
                       I  Starting a blocking GC Alloc
                       I  Waiting for a blocking GC Alloc
                       I  Background concurrent copying GC freed 876(191KB) AllocSpace objects, 15(75MB) LOS objects, 17% free, 113MB/137MB, paused 543us total 187.806ms
                       I  WaitForGcToComplete blocked Alloc on HeapTrim for 52.551ms
                       I  Starting a blocking GC Alloc
BARCODE                D  FPS: 25.74352 - Pool size: 8. Current analysis queue depth: 0. Current res: 3840*2160. Statistics are: {Point(3840, 2160)=3}
geodis.mobipre         I  Waiting for a blocking GC Alloc
                       I  Background concurrent copying GC freed 679(177KB) AllocSpace objects, 18(55MB) LOS objects, 15% free, 135MB/159MB, paused 3.672ms total 287.187ms
                       I  WaitForGcToComplete blocked Alloc on HeapTrim for 15.102ms
                       I  Starting a blocking GC Alloc
                       I  Waiting for a blocking GC Alloc
                       I  Background concurrent copying GC freed 955(190KB) AllocSpace objects, 13(66MB) LOS objects, 16% free, 118MB/142MB, paused 3.057ms total 212.534ms
                       I  WaitForGcToComplete blocked Alloc on HeapTrim for 87.613ms
                       I  Starting a blocking GC Alloc
                       I  Waiting for a blocking GC Alloc
                       I  Background concurrent copying GC freed 635(191KB) AllocSpace objects, 11(43MB) LOS objects, 14% free, 145MB/169MB, paused 11.419ms total 250.404ms
                       I  WaitForGcToComplete blocked Alloc on HeapTrim for 67.957ms
                       I  Starting a blocking GC Alloc
[... cut ...]
                          	at com.enioka.scanner.camera.-$$Lambda$CameraBarcodeScanViewV2$VMyTEjvJxZ7LD_CakOd_WrWwxVc.onImageAvailable(Unknown Source:2)
                          	at android.media.ImageReader$ListenerHandler.handleMessage(ImageReader.java:798)
                          	at android.os.Handler.dispatchMessage(Handler.java:107)
                          	at android.os.Looper.loop(Looper.java:237)
                          	at android.os.HandlerThread.run(HandlerThread.java:67)

```

abelatar, 2023-07-12:
J'ajoute encore ces traces qui bloquent la lecture par caméra :

```
Camera loop start has failed, this is usually due to changing resolution too fast. Error was: Session has been closed; further changes are illegal.
                                                                                                    java.lang.IllegalStateException: Session has been closed; further changes are illegal.
                                                                                                    	at android.hardware.camera2.impl.CameraCaptureSessionImpl.checkNotClosed(CameraCaptureSessionImpl.java:834)
                                                                                                    	at android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest(CameraCaptureSessionImpl.java:288)
                                                                                                    	at com.enioka.scanner.camera.CameraBarcodeScanViewV2$3.onConfigured(CameraBarcodeScanViewV2.java:586)
                                                                                                    	at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigured$0$CallbackProxies$SessionStateCallbackProxy(CallbackProxies.java:53)
                                                                                                    	at android.hardware.camera2.impl.-$$Lambda$CallbackProxies$SessionStateCallbackProxy$soW0qC12Osypoky6AfL3P2-TeDw.run(Unknown Source:4)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:883)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:100)
                                                                                                    	at android.os.Looper.loop(Looper.java:237)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2023-07-12 15:51:48.725 12826-31088 BARCODE                 com.geodis.mobicop                   D  Starting camera release
2023-07-12 15:51:48.725 12826-31088 BARCODE                 com.geodis.mobicop                   D   * Closing camera capture session
2023-07-12 15:51:48.725 12826-31088 BARCODE                 com.geodis.mobicop                   I  Camera repeating capture request was set up
2023-07-12 15:51:48.725 12826-31088 BARCODE                 com.geodis.mobicop                   I  Capture session is now configured and will start the capture request loop 248844473
2023-07-12 15:51:48.727 12826-31088 BARCODE                 com.geodis.mobicop                   D  Using metering zone (670,490) (770,590)
2023-07-12 15:51:48.728 12826-31088 BARCODE                 com.geodis.mobicop                   D  Setting AF mode to 4
2023-07-12 15:51:48.739 12826-31088 BARCODE                 com.geodis.mobicop                   I  Camera repeating capture request was set up
2023-07-12 15:51:48.739 12826-31088 BARCODE                 com.geodis.mobicop                   D  Capture session has nothing to process 248844473
2023-07-12 15:51:48.739 12826-31088 BARCODE                 com.geodis.mobicop                   I  Capture session has closed 232552724
2023-07-12 15:51:48.739 12826-31088 BARCODE                 com.geodis.mobicop                   D   * ImageReader closing
2023-07-12 15:51:48.739 12826-31088 BARCODE                 com.geodis.mobicop                   D   * Camera device closing
2023-07-12 15:51:48.805 12826-12826 Dialog                  com.geodis.mobicop                   D  mIsSamsungBasicInteraction = false
2023-07-12 15:51:48.805 12826-12826 Dialog                  com.geodis.mobicop                   D  mIsSamsungBasicInteraction = false, isMetaDataInActivity = false
2023-07-12 15:51:48.818 12826-12826 MultiWindowDecorSupport com.geodis.mobicop                   I  updateCaptionType >> DecorView@2de666a[Chargement], isFloating: true, isApplication: true, hasWindowDecorCaption: false, hasWindowControllerCallback: false
2023-07-12 15:51:48.818 12826-12826 MultiWindowDecorSupport com.geodis.mobicop                   D  setCaptionType = 0, DecorView = DecorView@2de666a[Chargement]
2023-07-12 15:51:48.832 12826-12826 ScrollView              com.geodis.mobicop                   D  initGoToTop
2023-07-12 15:51:48.838 12826-12826 ScrollView              com.geodis.mobicop                   D  initGoToTop
2023-07-12 15:51:48.859 12826-12826 ViewRootIm...hargement] com.geodis.mobicop                   D  setView = com.android.internal.policy.DecorView@2de666a TM=true MM=false
2023-07-12 15:51:48.875 12826-12826 ViewRootIm...sActivity] com.geodis.mobicop                   D  setView = com.android.internal.policy.DecorView@5a1d67c TM=true MM=false
2023-07-12 15:51:48.881 12826-12826 ScannerActivity         com.geodis.mobicop                   D  Status changed: CONNECTED --- Scanner connecté
2023-07-12 15:51:48.882 12826-12826 Choreographer           com.geodis.mobicop                   I  Skipped 38 frames!  The application may be doing too much work on its main thread.
2023-07-12 15:51:48.882 12826-12851 BufferQueueProducer     com.geodis.mobicop                   E  [ImageReader-1440x1080f23m16-12826-2] dequeueBuffer: BufferQueue has been abandoned
[... cut ...]
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1075)
```

marcanpilami, 2023-07-17:
Fixed in 59cac97313c52714eed49b0923469e8c2a7a756d

closed 2023-07-17 by marcanpilami