enioka-Haute-Couture/enioka_scan #103: Problème de commutation de mode caméra entre deux applications sur le TC27

abelatar, 2023-07-11:
Hello @marcanpilami 

Lorsque deux applications utilisent la bibliothèque de scan sur le même appareil TC27, la dernière application lancée est automatiquement forcée en mode caméra, même si l'initialisation de "goToCamera" est définie sur false dans le constructeur.

Étapes de reproduction :
Lancez une première application qui utilise la bibliothèque de scan sur le TC27.
Ensuite, lancez une deuxième application qui utilise également la bibliothèque de scan sur le même TC27.
Observez que la deuxième application est automatiquement forcée en mode caméra, même si l'initialisation de "goToCamera" est définie sur false dans le constructeur.

marcanpilami, 2023-08-21:
Hello ! C'est un comportement normal. En effet, l'activité `ScannerCompatActivity` cherche à faire en sorte qu'en cas de switch d'application puis retour il soit rapide de réutiliser les scanners. Pour ce faire, elle demande simplement une mise en pause.

C'est très important notamment pour les scanners BT, dont la séquence de connexion est très lente. Donc on reste connecté.

Cela a la conséquence que tu observe. Certains types de matériels, dont les Zebra EMDK, ont un lock global - une seule app a le droit d'utiliser le scanner en même temps. 

La solution est simple : si tu sais que tu va passer d'une app à l'autre (et dans ton cas c'est facile puisque c'est toi qui lance l'autre app depuis la première) il faut appeler `ScannerServiceApi.disconnect` avant de lancer l'autre app. Comme cela la première app abandonne complètement le scanner et la seconde pourra l'utiliser. Attention, si on revient à la première app ensuite le service sera déconnecté et il faudra relancer la connection aux scanners via `restartScannerDiscovery`. (mais c'est probablement en dehors de ton use case, je le mets surtout ici au cas où d'autres utilisateurs seraient intéressé par la solution).

abelatar, 2023-09-11:
Bonjour !
Merci pour la solution suggérée mais malgré avoir suivi les étapes fournies pour me déconnecter du scanner avant de lancer l'autre application, le problème persiste.