enioka-Haute-Couture/enioka_scan #111: Problème de scan avec la caméra lors d'ouverture d'une autre activité sur l'activité de scan

abelatar, 2023-09-04:
Hello @marcanpilami ,

La version 2.3.3 de la bibliothèque présente un problème de prévisualisation de la caméra et de scan, elle affiche un écran noir lorsqu'une nouvelle activité est ouverte sur l'activité de numérisation par caméra et on reviens à l'écran de numérisation.

[image]

Traces du Problème :

```

Buffer...ducer  E  [com.geodis.mobicop/com.geodis.mobicop.ScanDriverCodeActivity$_31184#0](id:ee60000018c,api:0,p:-1,c:3814) disconnect: not connected (req=1)
libEGL          W  EGLNativeWindowType 0xcaddf808 disconnect failed
SurfaceControl  I  nativeRelease 
![Screenshot_20230904_202412](https://github.com/enioka-Haute-Couture/enioka_scan/assets/94049129/0487b994-f5e3-4b9b-89c1-f6c8f93d7d1e)
nativeObject s[3079004800]
                I  nativeRelease nativeObject e[3079004800]
                I  assignNativeObject: nativeObject = 0 Surface(name=null)/@0x923aa9d / android.view.SurfaceControl.readFromParcel:1117 android.view.IWindowSession$Stub$Proxy.relayout:1810 android.view.ViewRootImpl.relayoutWindow:9005 android.view.ViewRootImpl.performTraversals:3360 android.view.ViewRootImpl.doTraversal:2618 android.view.ViewRootImpl$TraversalRunnable.run:9971 android.view.Choreographer$CallbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 
                I  assignNativeObject: nativeObject = 0 Surface(name=null)/@0xdeebc0c / android.view.SurfaceControl.readFromParcel:1117 android.view.IWindowSession$Stub$Proxy.relayout:1820 android.view.ViewRootImpl.relayoutWindow:9005 android.view.ViewRootImpl.performTraversals:3360 android.view.ViewRootImpl.doTraversal:2618 android.view.ViewRootImpl$TraversalRunnable.run:9971 android.view.Choreographer$CallbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 
                I  nativeRelease nativeObject s[3079005152]
                I  nativeRelease nativeObject e[3079005152]
ViewRo...vity]  I  Relayout returned: old=(0,0,720,1280) new=(0,0,720,1280) req=(720,1280)8 dur=9 res=0x5 s={false 0} ch=true fn=4
SurfaceView     I  windowStopped(true) false com.enioka.scanner.camera.CameraPreviewSurfaceView{22fe3e3 V.E...... ........ 0,0-720,952} of ViewRootImpl@9279274[ScanDriverCodeActivity]
BARCODE         D  Image reader already created
SurfaceControl  I  nativeRelease nativeObject s[3344440544]
                I  nativeRelease nativeObject e[3344440544]
ViewRo...vity]  I  MSG_WINDOW_FOCUS_CHANGED 1 1
InputM...nager  D  prepareNavigationBarInfo() DecorView@3091827[MessagesActivity]
                D  getNavigationBarColor() -855310
                D  prepareNavigationBarInfo() DecorView@3091827[MessagesActivity]
                D  getNavigationBarColor() -855310
                V  Starting input: tba=com.geodis.mobicop ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
                D  startInputInner - Id : 0
                I  startInputInner - mService.startInputOrWindowGainedFocus
InputTransport  D  Input channel constructed: 'ClientS', fd=111
                D  Input channel destroyed: 'ClientS', fd=73
libEGL          W  EGLNativeWindowType 0xcae43008 disconnect failed
SurfaceControl  I  nativeRelease nativeObject s[3312795744]
                I  nativeRelease nativeObject e[3312795744]
                I  assignNativeObject: nativeObject = 0 Surface(name=null)/@0x499ab12 / android.view.SurfaceControl.readFromParcel:1117 android.view.IWindowSession$Stub$Proxy.relayout:1810 android.view.ViewRootImpl.relayoutWindow:9005 android.view.ViewRootImpl.performTraversals:3360 android.view.ViewRootImpl.doTraversal:2618 android.view.ViewRootImpl$TraversalRunnable.run:9971 android.view.Choreographer$CallbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 
                I  assignNativeObject: nativeObject = 0 Surface(name=null)/@0x44c5c53 / android.view.SurfaceControl.readFromParcel:1117 android.view.IWindowSession$Stub$Proxy.relayout:1820 android.view.ViewRootImpl.relayoutWindow:9005 android.view.ViewRootImpl.performTraversals:3360 android.view.ViewRootImpl.doTraversal:2618 android.view.ViewRootImpl$TraversalRunnable.run:9971 android.view.Choreographer$CallbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 
ViewRo...lité]  I  Relayout returned: old=(18,493,702,835) new=(18,493,702,835) req=(684,342)8 dur=12 res=0x5 s={false 0} ch=true fn=128
Buffer...ducer  E  [com.geodis.mobicop/com.geodis.mobicop.WelcomeActivity$_31184#0](id:ee600000189,api:0,p:-1,c:3814) disconnect: not connected (req=1)
libEGL          W  EGLNativeWindowType 0xcade8408 disconnect failed
SurfaceControl  I  nativeRelease nativeObject s[3920229280]
                I  nativeRelease nativeObject e[3920229280]
                I  assignNativeObject: nativeObject = 0 Surface(name=null)/@0x97982e3 / android.view.SurfaceControl.readFromParcel:1117 android.view.IWindowSession$Stub$Proxy.relayout:1810 android.view.ViewRootImpl.relayoutWindow:9005 android.view.ViewRootImpl.performTraversals:3360 android.view.ViewRootImpl.doTraversal:2618 android.view.ViewRootImpl$TraversalRunnable.run:9971 android.view.Choreographer$CallbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 
                I  assignNativeObject: nativeObject = 0 Surface(name=null)/@0x14c949 / android.view.SurfaceControl.readFromParcel:1117 android.view.IWindowSession$Stub$Proxy.relayout:1820 android.view.ViewRootImpl.relayoutWindow:9005 android.view.ViewRootImpl.performTraversals:3360 android.view.ViewRootImpl.doTraversal:2618 android.view.ViewRootImpl$TraversalRunnable.run:9971 android.view.Choreographer$CallbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 
                I  nativeRelease nativeObject s[2999804320]
                I  nativeRelease nativeObject e[2999804320]
ViewRo...vity]  I  Relayout returned: old=(0,0,720,1280) new=(0,0,720,1280) req=(720,1280)8 dur=7 res=0x5 s={false 0} ch=true fn=6
SurfaceControl  I  nativeRelease nativeObject s[2999804384]
                I  nativeRelease nativeObject e[2999804384]
ViewRo...vity]  I  stopped(true) old=false
Gralloc4        W  allocator 3.x is not supported
Scanne...ivity  I  Scanner activity is being stopped 253460638
Gralloc3        W  allocator 3.x is not supported
ViewRo...lité]  I  stopped(true) old=false
ViewRo...vity]  I  stopped(true) old=false
                I  dispatchDetachedFromWindow
InputTransport  D  Input channel destroyed: 'a611e7c', fd=77
WindowManager   E  android.view.WindowLeaked: Activity com.geodis.mobicop.WelcomeActivity has leaked window DecorView@7ca4c3f[Vérification de compatibilité] that was originally added here
                   	at android.view.ViewRootImpl.<init>(ViewRootImpl.java:1049)
                   	at android.view.ViewRootImpl.<init>(ViewRootImpl.java:1023)
                   	at android.view.WindowManagerGlobal.addView(WindowManagerGlobal.java:450)
                   	at android.view.WindowManagerImpl.addView(WindowManagerImpl.java:114)
                   	at android.app.Dialog.show(Dialog.java:505)
                   	at android.app.ProgressDialog.show(ProgressDialog.java:225)
                   	at android.app.ProgressDialog.show(ProgressDialog.java:182)
                   	at android.app.ProgressDialog.show(ProgressDialog.java:167)
                   	at com.geodis.mobicop.WelcomeActivity.checkCompatibility(WelcomeActivity.java:365)
[... 186 lines cut ...]
                   	at java.lang.reflect.Method.invoke(Native Method)
                   	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
                   	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
BARCODE         D  Camera device not ready yet
ViewRo...vity]  I  MSG_WINDOW_FOCUS_CHANGED 1 1
InputM...nager  D  prepareNavigationBarInfo() DecorView@c2ac9b[ScanDriverCodeActivity]
                D  getNavigationBarColor() -855310
[... cut ...]
ViewRo...vity]  I  stopped(true) old=false
SurfaceControl  I  nativeRelease nativeObject s[2989577568]
                I  nativeRelease nativeObject e[2989577568]
Buffer...ducer  E  [com.geodis.mobicop/com.geodis.mobicop.MessagesActivity$_31184#0](id:ee60000018f,api:0,p:-1,c:3814) disconnect: not connected (req=1)
libEGL          W  EGLNativeWindowType 0xcae48408 disconnect failed
ViewRo...vity]  I  dispatchDetachedFromWindow
InputTransport  D  Input channel destroyed: '2e77260', fd=171
Camera...lobal  I  onTorchModeChanged

```

abelatar, 2023-09-25:
Le problème persiste sur la version 2.3.7 voici les straces : 

```
INFO =>         I  ScannerActivityStatus changed: CONNECTED --- Scanner connecté
ViewRo...vity]  D  MSG_WINDOW_FOCUS_CHANGED 1 1
InputM...nager  D  prepareNavigationBarInfo() DecorView@6f174f7[ScanDriverCodeActivity]
                D  getNavigationBarColor() -855310
                D  prepareNavigationBarInfo() DecorView@6f174f7[ScanDriverCodeActivity]
                D  getNavigationBarColor() -855310
                V  Starting input: tba=com.geodis.mobicop ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
                D  startInputInner - Id : 0
                I  startInputInner - mService.startInputOrWindowGainedFocus
InputTransport  D  Input channel destroyed: 'ClientS', fd=92
Camera...lobal  I  onTorchModeUnavailable
BARCODE         D  Trying to open camera from CameraManager 84459909
Camera...lobal  I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_IDLE for client com.geodis.mobicop API Level 2
AndroidRuntime  D  Shutting down VM
                E  FATAL EXCEPTION: main
                   Process: com.geodis.mobicop:acra, PID: 13186
                   java.lang.ClassCastException: android.os.BinderProxy cannot be cast to com.enioka.scanner.service.ScannerService$LocalBinder
                   	at com.geodis.mobicop.common.Mobicop$1.onServiceConnected(Mobicop.java:81)
                   	at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:2067)
                   	at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:2099)
                   	at android.os.Handler.handleCallback(Handler.java:883)
                   	at android.os.Handler.dispatchMessage(Handler.java:100)
                   	at android.os.Looper.loop(Looper.java:237)
                   	at android.app.ActivityThread.main(ActivityThread.java:7860)
                   	at java.lang.reflect.Method.invoke(Native Method)
                   	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
                   	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1075)
Camera...lobal  I  postSingleTorchUpdate device: camera id 0 status 1
                I  postSingleTorchUpdate device: camera id 0 status 1
                I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.geodis.mobicop API Level 2
                I  postSingleTorchUpdate device: camera id 0 status 0
                I  postSingleTorchUpdate device: camera id 0 status 0
                I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_OPEN for client com.geodis.mobicop API Level 2
BARCODE         D  Image reader already created
Camera...ssion  E  Session 0: Exception while stopping repeating: 
                   android.hardware.camera2.CameraAccessException: CAMERA_DISCONNECTED (2): cancelRequest:473: Camera device no longer alive
                   	at android.hardware.camera2.CameraManager.throwAsPublicException(CameraManager.java:1127)
                   	at android.hardware.camera2.impl.ICameraDeviceUserWrapper.cancelRequest(ICameraDeviceUserWrapper.java:97)
                   	at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:1151)
                   	at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:526)
                   	at android.hardware.camera2.impl.CameraCaptureSessionImpl$2.onDisconnected(CameraCaptureSessionImpl.java:737)
                   	at android.hardware.camera2.impl.CameraDeviceImpl$7.run(CameraDeviceImpl.java:244)
                   	at android.os.Handler.handleCallback(Handler.java:883)
                   	at android.os.Handler.dispatchMessage(Handler.java:100)
                   	at android.os.Looper.loop(Looper.java:237)
                   	at android.os.HandlerThread.run(HandlerThread.java:67)
                   Caused by: android.os.ServiceSpecificException: cancelRequest:473: Camera device no longer alive (code 4)
                   	at android.os.Parcel.createException(Parcel.java:2102)
                   	at android.os.Parcel.readException(Parcel.java:2056)
                   	at android.os.Parcel.readException(Parcel.java:2004)
                   	at android.hardware.camera2.ICameraDeviceUser$Stub$Proxy.cancelRequest(ICameraDeviceUser.java:676)
                   	at android.hardware.camera2.impl.ICameraDeviceUserWrapper.cancelRequest(ICameraDeviceUserWrapper.java:95)
                   	at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:1151) 
                   	at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:526) 
                   	at android.hardware.camera2.impl.CameraCaptureSessionImpl$2.onDisconnected(CameraCaptureSessionImpl.java:737) 
                   	at android.hardware.camera2.impl.CameraDeviceImpl$7.run(CameraDeviceImpl.java:244) 
                   	at android.os.Handler.handleCallback(Handler.java:883) 
                   	at android.os.Handler.dispatchMessage(Handler.java:100) 
                   	at android.os.Looper.loop(Looper.java:237) 
                   	at android.os.HandlerThread.run(HandlerThread.java:67) 
Choreographer   I  Skipped 35 frames!  The application may be doing too much work on its main thread.
[... cut ...] Callbacks.onDeviceError(int, android.hardware.camera2.impl.CaptureResultExtras) for 416ms
BARCODE         V  CameraDevice.StateCallback.onDisconnected 84459909
                V  CameraDevice.StateCallback.onOpened 84459909
                D  Image reader already created
                V  CameraDevice.StateCallback.onClosed 84459909
                D   * Background camera thread is closing
                D  Stopping background thread handler 84459909
                D   * Closing analyzers
```

abelatar, 2023-10-02:
Hello @marcanpilami 

Il y a encore ce problème:

```
vity]  I  Relayout returned: old=(0,0,720,1280) new=(0,0,720,1280) req=(720,1280)8 dur=11 res=0x5 s={false 0} ch=true fn=7
SurfaceControl  I  nativeRelease nativeObject s[3661104096]
                I  nativeRelease nativeObject e[3661104096]
ViewRo...vity]  I  stopped(true) old=false
Scanne...ivity  I  Scanner activity is being stopped 255049012
ViewRo...lité]  I  stopped(true) old=false
ViewRo...vity]  I  stopped(true) old=false
                I  dispatchDetachedFromWindow
InputTransport  D  Input channel destroyed: '7951db1', fd=74
WindowManager   E  android.view.WindowLeaked: Activity com.geodis.mobicop.WelcomeActivity has leaked window DecorView@86d4319[Vérification de compatibilité] that was originally added here
                   	at android.view.ViewRootImpl.<init>(ViewRootImpl.java:1049)
                   	at android.view.ViewRootImpl.<init>(ViewRootImpl.java:1023)
                   	at android.view.WindowManagerGlobal.addView(WindowManagerGlobal.java:450)
                   	at android.view.WindowManagerImpl.addView(WindowManagerImpl.java:114)
                   	at android.app.Dialog.show(Dialog.java:505)
                   	at android.app.ProgressDialog.show(ProgressDialog.java:225)
                   	at android.app.ProgressDialog.show(ProgressDialog.java:182)
                   	at android.app.ProgressDialog.show(ProgressDialog.java:167)
                   	at com.geodis.mobicop.WelcomeActivity.checkCompatibility(WelcomeActivity.java:364)
                   	at com.geodis.mobicop.WelcomeActivity.onSplashClick(WelcomeActivity.java:312)
                   	at java.lang.reflect.Method.invoke(Native Method)
                   	at androidx.appcompat.app.AppCompatViewInflater$DeclaredOnClickListener.onClick(AppCompatViewInflater.java:441)
                   	at android.view.View.performClick(View.java:8160)
                   	at android.widget.TextView.performClick(TextView.java:16222)
                   	at android.view.View.performClickInternal(View.java:8137)
                   	at android.view.View.access$3700(View.java:888)
                   	at android.view.View$PerformClick.run(View.java:30236)
                   	at android.os.Handler.handleCallback(Handler.java:938)
                   	at android.os.Handler.dispatchMessage(Handler.java:99)
                   	at android.os.Looper.loop(Looper.java:246)
                   	at android.app.ActivityThread.main(ActivityThread.java:8645)
                   	at java.lang.reflect.Method.invoke(Native Method)
                   	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
                   	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
                E  win=Window{7a01433 u0 com.geodis.mobicop/com.geodis.mobicop.ScanDriverCodeActivity} destroySurfaces: appStopped=true win.mWindowRemovalAllowed=false win.mRemoveOnExit=false win.mViewVisibility=8 caller=com.android.server.wm.ActivityRecord.destroySurfaces:5251 com.android.server.wm.ActivityRecord.destroySurfaces:5232 com.android.server.wm.ActivityRecord.notifyAppStopped:5291 com.android.server.wm.ActivityRecord.activityStopped:5918 com.android.server.wm.ActivityTaskManagerService.activityStopped:2595 android.app.IActivityTaskManager$Stub.onTransact:2223 android.os.Binder.execTransactInternal:1190 
ViewRo...lité]  I  dispatchDetachedFromWindow
InputTransport  D  Input channel destroyed: '265e6c7', fd=87
Buffer...ducer  E  [SurfaceView - com.geodis.mobicop/com.geodis.mobicop.ScanDriverCodeActivity@ef90b3a@0#0](id:ee80000014a,api:4,p:3864,c:3816) dequeueBuffer: BufferQueue has been abandoned
BARCODE         D  Capture session has nothing to process 58261751
AudioC...ities  W  Unsupported mime audio/x-ima
VideoC...ities  W  Unsupported mime video/mp43
                W  Unsupported mime video/wvc1
                W  Unsupported mime video/x-ms-wmv
AudioC...ities  W  Unsupported mime audio/x-ms-wma
VideoC...ities  W  Unsupported mime video/x-ms-wmv7
                W  Unsupported mime video/x-ms-wmv8
AudioC...ities  W  Unsupported mime audio/x-ima
                W  Unsupported mime audio/mpeg-L1
                W  Unsupported mime audio/mpeg-L2
VideoC...ities  W  Unsupported mime video/wvc1
                W  Unsupported mime video/wvc1
                W  Unsupported mime video/avc-wfd
                W  Unrecognized profile/level 1/32 for video/mp4v-es
                W  Unrecognized profile/level 32768/2 for video/mp4v-es
                W  Unrecognized profile/level 32768/64 for video/mp4v-es
ViewRo...vity]  I  ViewPostIme pointer 0
                I  ViewPostIme pointer 1
Messag...ivity  D  clickNext() called with: v = [androidx.appcompat.widget.AppCompatButton{7fa89c1 VFED..C.. ...P.... 4,4-676,100 #7f080084 app:id/btn_next}]
ViewRo...vity]  I  ViewPostIme pointer 0
                I  ViewPostIme pointer 1
Messag...ivity  D  clickFinish() called with: v = [androidx.appcompat.widget.AppCompatButton{87cb1f2 VFED..C.. ...P.... 344,4-676,100 #7f080083 app:id/btn_finish}]
ViewRo...vity]  I  stopped(false) old=true
Scanne...ivity  D  Scanner activity is starting 255049012
ViewRo...vity]  I  stopped(false) old=false
Scanne...ivity  I  Scanner activity is resuming 255049012
[... 130 lines cut ...]
                   	at android.os.Handler.dispatchMessage(Handler.java:99)
                   	at android.os.Looper.loop(Looper.java:246)
[... cut ...]
                D  Input channel destroyed: 'ClientS', fd=207
ViewRo...vity]  I  stopped(true) old=false
SurfaceControl  I  nativeRelease nativeObject s[3161790400]
                I  nativeRelease nativeObject e[3161790400]
Buffer...ducer  E  [com.geodis.mobicop/com.geodis.mobicop.MessagesActivity$_1683#0](id:ee80000014c,api:0,p:-1,c:3816) disconnect: not connected (req=1)
libEGL          W  EGLNativeWindowType 0xf0fe5808 disconnect failed
ViewRo...vity]  I  dispatchDetachedFromWindow
InputTransport  D  Input channel destroyed: '21f01b9', fd=177
```

abelatar, 2023-10-06:
Hello @marcanpilami ,
Sur cette version, l'application s'arrête complètement avec une exception liée à la caméra. Voici les traces d'erreur :
```
      E  FATAL EXCEPTION: main
                   Process: com.geodis.mobicop, PID: 2269
                   java.lang.IllegalStateException: cannot acquire camera semaphore - it may be locked by another activity/fragment
                   	at com.enioka.scanner.camera.CameraBarcodeScanViewV2.acquire(CameraBarcodeScanViewV2.java:805)
                   	at com.enioka.scanner.camera.CameraBarcodeScanViewV2.openCamera(CameraBarcodeScanViewV2.java:272)
                   	at com.enioka.scanner.camera.CameraBarcodeScanViewV2.lambda$6PHodskDk4Q6UmexosXlEBZZev4(Unknown Source:0)
                   	at com.enioka.scanner.camera.-$$Lambda$CameraBarcodeScanViewV2$6PHodskDk4Q6UmexosXlEBZZev4.run(Unknown Source:2)
                   	at android.os.Handler.handleCallback(Handler.java:938)
                   	at android.os.Handler.dispatchMessage(Handler.java:99)
                   	at android.os.Looper.loop(Looper.java:246)
                   	at android.app.ActivityThread.main(ActivityThread.java:8645)
                   	at java.lang.reflect.Method.invoke(Native Method)
                   	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
                   	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
---------------------------- PROCESS STARTED (4411) for package com.geodis.mobicop ----------------------------
ScannerService  I  Starting scanner service
                I  ScannerService is receiving a new bind request - onBind
                I  (re)starting provider discovery!
LaserScanner    I  Starting provider discovery
                I  There are 7 scanner provider(s) available before filtering duplicates
                D  Trying to instantiate provider com.enioka.scanner.sdk.zebra.dw.ZebraDwProvider - not BT - 0
                D  Provider ZebraDwProvider was successfully instantiated
                D  Trying to instantiate provider com.enioka.scanner.sdk.athesi.SPA43LTE.AthesiHHTProvider - not BT - 0
                D  Provider AthesiHHTProvider was successfully instantiated
                D  Trying to instantiate provider com.enioka.scanner.sdk.athesi.RD50TE.AthesiE5LProvider - not BT - 0
                D  Provider AthesiE5LProvider was successfully instantiated
                D  Trying to instantiate provider com.enioka.scanner.sdk.bluebird.BluebirdProvider - not BT - 0
                D  Provider BluebirdProvider was successfully instantiated
                D  Trying to instantiate provider com.enioka.scanner.sdk.honeywelloss.integrated.HoneywellOssIntegratedScannerProvider - not BT - 0
                D  Provider HoneywellOssIntegratedProvider was successfully instantiated
                D  Trying to instantiate provider com.enioka.scanner.sdk.proglove.ProgloveProvider - BT     - 0
                D  Provider ProgloveProvider was successfully instantiated
                D  Trying to instantiate provider com.enioka.scanner.bt.manager.SerialBtScannerProvider - BT     - 0
                D  Provider BtSppSdk was successfully instantiated
BtSppSdk        I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.generalscan.GsSppScannerProvider
                I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.zebraoss.ZebraOssSppScannerProvider
                I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.zebraoss.ZebraOssAttScannerProvider
                I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.honeywelloss.spp.HoneywellOssSppScannerProvider
LaserScanner    I  Provider discovery done
ScannerService  D  Status changed: SERVICE_PROVIDER_SEARCH_OVER --- ScannerService a terminé la découverte de providers de scanner
                I  (re)starting scanner search!
LaserScanner    I  Starting scanner search
                D  Provider BluebirdProvider accepted
                D  Provider ZebraDwProvider accepted
                D  Provider AthesiHHTProvider accepted
                D  Provider HoneywellOssIntegratedProvider accepted
                D  Provider BtSppSdk skipped because bluetooth option is disabled
                D  Provider AthesiE5LProvider accepted
                D  Provider ProgloveProvider skipped because bluetooth option is disabled
                I  There are 5 providers which are going to be invoked for fresh laser scanners
                I  	ZebraDwProvider
                I  	HoneywellOssIntegratedProvider
                I  	BluebirdProvider
                I  	AthesiHHTProvider
                I  	AthesiE5LProvider
                I  Starting search on provider ZebraDwProvider
                I  Starting search on provider HoneywellOssIntegratedProvider
                I  Starting search on provider BluebirdProvider
                I  Scanner provider ZebraDwProvider reports it is not compatible with the device and will be disabled
                I  Starting search on provider AthesiHHTProvider
                I  Scanner provider AthesiHHTProvider reports it is not compatible with the device and will be disabled
                I  Starting search on provider AthesiE5LProvider
                I  Scanner provider AthesiE5LProvider reports it is not compatible with the device and will be disabled
                I  Scanner provider BluebirdProvider reports it is not compatible with the device and will be disabled
[... cut ...]
                   	at android.os.Looper.loop(Looper.java:246)
                   	at android.app.ActivityThread.main(ActivityThread.java:8645)
                   	at java.lang.reflect.Method.invoke(Native Method)
                   	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
                   	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
---------------------------- PROCESS ENDED (4411) for package com.geodis.mobicop ----------------------------
```

DaSpood, 2023-10-06:
Hello @abelatar , cette erreur ne doit se produire que lorsque plusieurs appels a `openCamera()` (donc `surfaceCreated()`) sont effectués, donc plusieurs instances de la camera sont ouvertes dans la meme application sans être fermées entre chaque appel.

La camera se ferme en cas d'erreur, d'appel a `surfaceDestroyed()`, ou d'appel a la fonction `cleanUp()` de la vue camera, elle meme appelable par la fonction `disconnect()` du scanner camera.

N'y a-t-il bien qu'une seule instance de la vue ou du scanner camera dans l'application ?

abelatar, 2023-10-06:
J'ai supprimé la contrainte de mode caméra pour les téléphones autres que le TC27, et cela fonctionne correctement. Voici le code associé :
`this.goToCamera = deviceName == null || !deviceName.equalsIgnoreCase(Mobicop.TC27);
`

Dois-je retirer ce code de toutes mes activités ? Est-ce désormais la bibliothèque qui décide si on passe en mode scanner ou caméra sur les versions récentes ?

DaSpood, 2023-10-06:
Hello @abelatar 

Actuellement, la bibliothèque passe automatiquement en mode camera si aucun autre scanner n'est disponible, et passe en mode "classique" si au moins un scanner est disponible.

Le champ `goToCamera` est set a `true` lors de l'appel a la fonction `initCamera()`, qui est appelée soit manuellement, soit automatiquement a la fin de la recherche de scanners si aucun scanner n'est disponible.