enioka-Haute-Couture/enioka_scan #113: Problème de configuration des profils DataWedge avec la version 2.3.7

abelatar, 2023-09-21:
Hello @marcanpilami 
Après avoir mis à jour à la version 2.3.7, je constate un souci avec la configuration des profils DataWedge. Seul le Profile0 semble fonctionner correctement. Les autres profils configurés ne sont pas reconnus ou ne fonctionnent pas comme prévu.

De plus, si je configure un profil nommé "geodis" et que je désactive les autres profils, le système semble toujours rechercher Profile0 plutôt que mon profil "geodis". En conséquence, le scan ne se déclenche pas lorsque Profile0 est désactivé il faut revoir qlqchose côté lib ? 

```
Start proc 15906:com.geodis.mobiprep/u0a269 for next-top-activity {com.geodis.mobiprep/com.geodis.mobiprep.WelcomeActivity}
18:31:59.695 ContextImpl                   W  Calling a method in the system process without a qualified user: android.app.ContextImpl.sendBroadcast:1189 com.qualcomm.qti.Performance$SendGameLaunchBroadcast.run:247 java.lang.Thread.run:1012 <bottom of call stack> <bottom of call stack> 
18:31:59.702 f                             D  componentInfo - com.geodis.mobiprep.WelcomeActivity
18:31:59.702                               D  getForegroundActivity changed
18:31:59.702 ActivityChangeNotifier        D  onActivityChange
18:31:59.703                               D  getCurrentActivity
18:31:59.704 TAG                           D  Process Activity Change
18:31:59.704 ActivityTracker               D  #ACT# Handle New activity, p= com.geodis.mobiprep a=com.geodis.mobiprep.WelcomeActivity
18:31:59.706                               D  loading active profile 1 [start]
18:31:59.708 DWHelperService               D  Using app context
18:31:59.708 tacapturevision               W  Large reply transaction of 410424 bytes, interface descriptor com.zebra.dcvision.IDCVisionInfoProxy, code 1
18:31:59.710 RFIDAPI3Error                 D  RfidUsbMgr: connection failed: device not found
18:31:59.712 SQLiteLog                     W  (28) double-quoted string literal: "com.geodis.mobiprep"
18:31:59.720 audio_hw_primary              D  start_output_stream: stream_handle(0xe64821a0) usecase(1: low-latency-playback) out_snd_device(2: speaker)             acdb_id(15) sample_rate(48000) channel_mask(0x3) app_type(69937)             audio_mode(0 : MODE_NORMAL) out_flags(0x6 : OUTPUT_FLAG_PRIMARY|OUTPUT_FLAG_FAST) out_devices(0x2 : OUT_SPEAKER) 
18:31:59.720                               D  start_output_stream: exit
18:31:59.723 geodis.mobiprep               I  Late-enabling -Xcheck:jni
18:31:59.735 ActivityTracker               D  updateCurrentProfile loadActiveProfile
18:31:59.735                               D  updating current profile: id: 1, name: Profile0 (default)
18:31:59.735                               D  disableCurrentProfile next profile disabled
18:31:59.735 ScanningProfile               D  #DWProfile# Disable profile Launcher [Start]
18:31:59.735 DWIMSHelper                   D  resetIMS()
18:31:59.736 DWHelper                      D  getHelper
18:31:59.736 c                             D  getIMSHelper
18:31:59.741 h                             D  resetIMS()
18:31:59.741 WorkFlow-Plugin               D  #DCVision# disable took 0
18:31:59.741 ScanningProfile               D  #DWProfile# Disable profile Launcher [Finish] took 6 ms
18:31:59.741 BroadcastHelper               D  Sending broadcast to handler com.symbol.datawedge.api.NOTIFICATION_ACTION
18:31:59.742                               D  Sending broadcast intent com.symbol.datawedge.api.NOTIFICATION_ACTION
18:31:59.742 DWHelper                      D  getHelper
18:31:59.743 BroadcastHelper               D  Broadcast intent finished com.symbol.datawedge.api.NOTIFICATION_ACTION
18:31:59.743 APINotificationManager        D  #DWProfile# Sent Profile Switch Notification Intent for profile: Profile0 (default)
18:31:59.743 ActivityTracker               D  loading active profile 1 [finish]
18:31:59.750 adbd                          I  jdwp connection from 15906
18:31:59.771 ActivityManager               E  Sending non-protected broadcast com.symbol.datawedge.api.NOTIFICATION_ACTION from system 2637:com.zebra.zvahelper/u0a99 pkg com.zebra.zvahelper
                                              java.lang.Throwable
                                              	at com.android.server.am.ActivityManagerService.checkBroadcastFromSystem(ActivityManagerService.java:13670)
                                              	at com.android.server.am.ActivityManagerService.broadcastIntentLocked(ActivityManagerService.java:14297)
                                              	at com.android.server.am.ActivityManagerService.broadcastIntentLocked(ActivityManagerService.java:13688)
                                              	at com.android.server.am.ActivityManagerService.broadcastIntentWithFeature(ActivityManagerService.java:14566)
                                              	at android.app.IActivityManager$Stub.onTransact(IActivityManager.java:2296)
                                              	at com.android.server.am.ActivityManagerService.onTransact(ActivityManagerService.java:2654)
                                              	at android.os.Binder.execTransactInternal(Binder.java:1280)
                                              	at android.os.Binder.execTransact(Binder.java:1244)
18:31:59.772                               E  Sending non-protected broadcast com.symbol.datawedge.api.NOTIFICATION_ACTION from system 2637:com.zebra.zvahelper/u0a99 pkg com.zebra.zvahelper
                                              java.lang.Throwable
                                              	at com.android.server.am.ActivityManagerService.checkBroadcastFromSystem(ActivityManagerService.java:13670)
                                              	at com.android.server.am.ActivityManagerService.broadcastIntentLocked(ActivityManagerService.java:14392)
                                              	at com.android.server.am.ActivityManagerService.broadcastIntentLocked(ActivityManagerService.java:13688)
                                              	at com.android.server.am.ActivityManagerService.broadcastIntentWithFeature(ActivityManagerService.java:14566)
                                              	at android.app.IActivityManager$Stub.onTransact(IActivityManager.java:2296)
                                              	at com.android.server.am.ActivityManagerService.onTransact(ActivityManagerService.java:2654)
                                              	at android.os.Binder.execTransactInternal(Binder.java:1280)
                                              	at android.os.Binder.execTransact(Binder.java:1244)
18:31:59.776 CompatibilityInfo             D  mCompatibilityFlags - 0
18:31:59.776                               D  applicationDensity - 480
18:31:59.776                               D  applicationScale - 1.0
18:31:59.778                               D  mCompatibilityFlags - 0
18:31:59.778                               D  applicationDensity - 480
18:31:59.778                               D  applicationScale - 1.0
18:31:59.778 AppsFilter                    I  interaction: PackageSetting{33defd1 com.geodis.mobiprep/10269} -> PackageSetting{6b41982 com.android.launcher3/10199} BLOCKED
18:31:59.786 JobScheduler                  I  bypass checkIfRestricted for JobStatus{11e77e #u0a53/900 com.symbol.fota/.jobservices.CTLAwakeService u=0 s=10053 TIME=+14m27s211ms:+14m36s211ms satisfied:0xb400000 unsatisfied:0x80000000}
18:31:59.786                               I  bypass checkIfRestricted for JobStatus{11e77e #u0a53/900 com.symbol.fota/.jobservices.CTLAwakeService u=0 s=10053 TIME=+14m27s211ms:+14m36s211ms satisfied:0xb400000 unsatisfied:0x80000000}
18:31:59.788 ActivityThread                I  Relaunch all activities: onCoreSettingsChange
---------------------------- PROCESS STARTED (15906) for package com.geodis.mobiprep ----------------------------
[... 40 lines cut ...]
18:32:00.139                               D  Trying to instantiate provider com.enioka.scanner.sdk.zebra.dw.ZebraDwProvider - not BT - 0
18:32:00.140                               D  Provider ZebraDwProvider was successfully instantiated
18:32:00.140                               D  Trying to instantiate provider com.enioka.scanner.sdk.athesi.SPA43LTE.AthesiHHTProvider - not BT - 0
18:32:00.140                               D  Provider AthesiHHTProvider was successfully instantiated
[... cut ...]
18:32:00.144                               I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.zebraoss.ZebraOssSppScannerProvider
18:32:00.144                               I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.zebraoss.ZebraOssAttScannerProvider
18:32:00.145                               I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.honeywelloss.spp.HoneywellOssSppScannerProvider
18:32:00.145 LaserScanner                  I  Provider discovery done
18:32:00.147 ActivityTaskManager           I  The Process com.geodis.mobiprep Already Exists in BG. So sending its PID: 15906
```

marcanpilami, 2023-10-01:
We've clarified profile creation in the doc, and to further make things easier we now create a profile. See [the doc for more info](https://github.com/enioka-Haute-Couture/enioka_scan/blob/mag/geodis/enioka_scan_zebra_dw/README.md)

abelatar, 2023-10-05:
The scanning issue arises when the app is placed in the background with profile0 desabled. Upon return, the scan works the first time but deactivates on the second attempt. 
I've imported this dependency : implementation 'com.enioka.scanner:provider-cs-zebra-dw:2.3.8'.

Steps to reproduce:

1 - Open any screen in the app that supports laser scanning.
2 - Activate the laser scanning feature.
3 - Confirm that the laser scanning functions correctly.
4 - Place the app in the background (don't fully close it).
5 - Return to the app.
6 - Attempt a laser scan.
7 - Note whether the scan works or not (in my case, it doesn't).
8 - Repeat from step 4

```
                D  Scanner activity is starting 207974906
                I  Scanner activity is resuming 207974906
                I  Resuming scanner activity - scanners will be (re)connected
ZebraD...vider  D  Sending intent to scanner to enable the trigger
ScannerService  D  Registering new client: com.geodis.mobicop.ScanDriverCodeActivity@c6571fa 207974906
                D  Notifying late clients that providers are already discovered
                D  Notifying late clients that scanners are already connected
Scanne...ivity  I  Activity can now use all received scanners (1)
ZebraDwScanner  D  Received data from DW service
                I  PROFILE_SWITCH: profileName: Profile0 (default), profileEnabled: false
                D  Sending DW action. DW command ID: 29539A793B954065A8E8B438D1E95E68	Intent action: com.symbol.datawedge.api.ACTION
                   	com.symbol.datawedge.api.SWITCH_SCANNER_PARAMS=Bundle[{decoder_ean8=false, decoder_upca=false, decoder_aztec=false, decoder_upce0=false, decoder_mailmark=false, decoder_maxicode=false, decoder_codabar=false, decoder_datamatrix=false, decoder_code39=false, decoder_pdf417=false, decoder_qrcode=false}]
                   	SEND_RESULT=true
                   	COMMAND_IDENTIFIER=29539A793B954065A8E8B438D1E95E68
                D  Received data from DW service
                D  Received result for datawedge command com.symbol.datawedge.api.SWITCH_SCANNER_PARAMS -  FAILURE - 29539A793B954065A8E8B438D1E95E68
                E  Zebra DataWedge command failure: PROFILE_NAME = Profile0 (default)
                   RESULT_CODE = [Ljava.lang.String;@af65da6
                   #Intent;action=com.symbol.datawedge.api.RESULT_ACTION;category=android.intent.category.DEFAULT;launchFlags=0x10;S.RESULT=FAILURE;S.COMMAND_IDENTIFIER=29539A793B954065A8E8B438D1E95E68;S.COMMAND=com.symbol.datawedge.api.SWITCH_SCANNER_PARAMS;end
Scanne...ivity  I  Scanner activity is being paused 207974906
ZebraD...vider  D  Sending intent to scanner to disable the trigger
Scanne...ivity  I  Scanner activity is being stopped 207974906
                D  Scanner activity is starting 207974906
                I  Scanner activity is resuming 207974906
                I  Resuming scanner activity - scanners will be (re)connected
ZebraD...vider  D  Sending intent to scanner to enable the trigger
ZebraDwScanner  D  Sending DW action. DW command ID: 449AB912310F48C58D76EAF8EB34D360	Intent action: com.symbol.datawedge.api.ACTION
                   	com.symbol.datawedge.api.SWITCH_TO_PROFILE=com.geodis.mobicop
                   	SEND_RESULT=true
                   	COMMAND_IDENTIFIER=449AB912310F48C58D76EAF8EB34D360
ScannerService  D  Registering new client: com.geodis.mobicop.ScanDriverCodeActivity@c6571fa 207974906
                D  Notifying late clients that providers are already discovered
                D  Notifying late clients that scanners are already connected
Scanne...ivity  I  Activity can now use all received scanners (1)
ZebraDwScanner  D  Received data from DW service
                D  Received result for datawedge command com.symbol.datawedge.api.SWITCH_TO_PROFILE -  SUCCESS - 449AB912310F48C58D76EAF8EB34D360
                D  Profile switch done with success
                D  Received data from DW service
                I  PROFILE_SWITCH: profileName: com.geodis.mobicop, profileEnabled: true
                D  Sending DW action. DW command ID: E08031B23190416A8A58CC9D526A761D	Intent action: com.symbol.datawedge.api.ACTION
                   	com.symbol.datawedge.api.SWITCH_SCANNER_PARAMS=Bundle[{decoder_ean8=false, decoder_upca=false, decoder_aztec=false, decoder_upce0=false, decoder_mailmark=false, decoder_maxicode=false, decoder_codabar=false, decoder_datamatrix=false, decoder_code39=false, decoder_pdf417=false, decoder_qrcode=false}]
                   	SEND_RESULT=true
                   	COMMAND_IDENTIFIER=E08031B23190416A8A58CC9D526A761D
                D  Received data from DW service
                D  Received result for datawedge command com.symbol.datawedge.api.SWITCH_SCANNER_PARAMS -  SUCCESS - E08031B23190416A8A58CC9D526A761D
                I  DW action is done - configured {decoder_upca=false, decoder_pdf417=false, decoder_aztec=false, decoder_codabar=false, decoder_datamatrix=false, decoder_ean8=false, decoder_qrcode=false, decoder_mailmark=false, decoder_maxicode=false, decoder_code39=false, decoder_upce0=false}
                D  Received data from DW service
                D  SCANNER_STATUS: status: IDLE, profileName: com.geodis.mobicop
ScannerService  D  Status changed: INITIALIZING --- Scanner is in the process of initializing
Scanne...ivity  D  Status changed: INITIALIZING --- Scanner is in the process of initializing
ZebraDwScanner  D  Received data from DW service
[... cut ...]
                D  Received result for datawedge command com.symbol.datawedge.api.SWITCH_SCANNER_PARAMS -  FAILURE - E26388872F56471BAB9F852AC500986F
                E  Zebra DataWedge command failure: PROFILE_NAME = Profile0 (default)
                   RESULT_CODE = [Ljava.lang.String;@fb79cd5
                   #Intent;action=com.symbol.datawedge.api.RESULT_ACTION;category=android.intent.category.DEFAULT;launchFlags=0x10;S.RESULT=FAILURE;S.COMMAND_IDENTIFIER=E26388872F56471BAB9F852AC500986F;S.COMMAND=com.symbol.datawedge.api.SWITCH_SCANNER_PARAMS;end
```

ffayel, 2023-10-05:
@marcanpilami  : le sujet et toujours d'actualité. Peut-on le rouvrir ?

commit 425cc187 (2023-10-01): Closes #113 - Zebra DW provider now creates its own profile.
commit 9ea58b36 (2023-10-05): Fixed #113 remark - DW profile switching.
commit 1bae8723 (2023-10-06): Closes #113 - Zebra DW provider now creates its own profile.
commit 7126f927 (2023-10-06): Add Zebra DataWedge provider (#126)

* Added Zebra DW provider
* Fixed #112 : worked around camera lifecycle.
* Fixed DW provider start/stop API
* Fixed incomplete CameraV2 image
* Activity: Fixed LED button always present in default layout
* Closes #113 - Zebra DW provider now creates its own profile.
* Fixed API version
* Fixed API 24 call
* Fixed #124: IntentProvider service test
* Fixed #124: ZebraDwProvider compatibility test
* Add <queries> tag to manifest for DW package
* Tentative fix for #111 - camera init/close is now locked globally
* Fixed #113 remark - DW profile switching.

----- [... cut]
closed 2023-10-06 by DaSpood