enioka-Haute-Couture/enioka_scan #124: Mode scanner activé sur un Samsung Xcover 4

abelatar, 2023-10-03:
Lors du démarrage de Mobicop sur un Samsung Xcover 4, la bibliothèque démarre en mode scanner, bien que cet appareil ne devrait pas prendre en charge ce mode.

**Comportement attendu :**
L'application devrait reconnaître que l'appareil ne prend pas en charge le mode scanner et ne pas essayer de l'initialiser.

 **Traces :** 

```
getUsableFileNum - File num : 0
                D  DataStore created
                V  Loading metadata form file
                V  Datastore version : 1
Journe...State  D  new and empty journey state
Mobico...ation  D  startTrackingLocation Location 
art             W  Before Android 4.1, method android.graphics.PorterDuffColorFilter androidx.vectordrawable.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
TextView        D  setTypeface with style : 0
                D  setTypeface with style : 0
                D  setTypeface with style : 0
                D  setTypeface with style : 0
ViewRo...vity]  D  ThreadedRenderer.create() translucent=false
InputTransport  D  Input channel constructed: fd=59
ViewRo...vity]  D  setView = DecorView@588c656[WelcomeActivity] touchMode=true
ScannerService  I  Starting scanner service
                I  ScannerService is receiving a new bind request - onBind
                I  (re)starting provider discovery!
LaserScanner    I  Starting provider discovery
                I  There are 15 scanner provider(s) available before filtering duplicates
                D  Trying to instantiate provider com.enioka.scanner.sdk.zebra.dw.ZebraDwProvider - not BT - 0
                D  Provider ZebraDwProvider was successfully instantiated
                D  Trying to instantiate provider com.enioka.scanner.sdk.athesi.SPA43LTE.AthesiHHTProvider - not BT - 0
                D  Provider AthesiHHTProvider was successfully instantiated
                D  Trying to instantiate provider com.enioka.scanner.sdk.athesi.RD50TE.AthesiE5LProvider - not BT - 0
                D  Provider AthesiE5LProvider was successfully instantiated
                D  Trying to instantiate provider com.enioka.scanner.sdk.bluebird.BluebirdProvider - not BT - 0
                D  Provider BluebirdProvider was successfully instantiated
                D  Trying to instantiate provider com.enioka.scanner.sdk.honeywelloss.integrated.HoneywellOssIntegratedScannerProvider - not BT - 0
                D  Provider HoneywellOssIntegratedProvider was successfully instantiated
                D  Trying to instantiate provider com.enioka.scanner.sdk.proglove.ProgloveProvider - BT     - 0
                D  Provider ProgloveProvider was successfully instantiated
                D  Trying to instantiate provider com.enioka.scanner.bt.manager.SerialBtScannerProvider - BT     - 0
                D  Provider BtSppSdk was successfully instantiated
BtSppSdk        I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.generalscan.GsSppScannerProvider
                I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.zebraoss.ZebraOssSppScannerProvider
                I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.zebraoss.ZebraOssAttScannerProvider
                I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.honeywelloss.spp.HoneywellOssSppScannerProvider
                D  Skipping duplicate provider com.enioka.scanner.sdk.generalscan.GsSppScannerProvider : does not match application UID (Service=10156 | Application=10167)
                D  Skipping duplicate provider com.enioka.scanner.sdk.zebraoss.ZebraOssSppScannerProvider : does not match application UID (Service=10156 | Application=10167)
                D  Skipping duplicate provider com.enioka.scanner.sdk.honeywelloss.HoneywellOssSppScannerProvider : does not match application UID (Service=10156 | Application=10167)
LaserScanner    D  Skipping duplicate provider com.enioka.scanner.sdk.hid.GenericHidProvider : does not match application UID (Service=10156 | Application=10167)
                D  Skipping duplicate provider com.enioka.scanner.sdk.athesi.HHTProvider : does not match application UID (Service=10156 | Application=10167)
                D  Skipping duplicate provider com.enioka.scanner.sdk.bluebird.BluebirdProvider : does not match application UID (Service=10156 | Application=10167)
                D  Skipping duplicate provider com.enioka.scanner.sdk.proglove.ProgloveProvider : does not match application UID (Service=10156 | Application=10167)
                D  Skipping duplicate provider com.enioka.scanner.bt.manager.SerialBtScannerProvider : does not match application UID (Service=10156 | Application=10167)
                D  Skipping duplicate provider com.enioka.scanner.sdk.koamtac.KoamtacScannerProvider : does not match application UID (Service=10156 | Application=10167)
                D  Skipping duplicate provider com.enioka.scanner.sdk.zebra.BtZebraProvider : does not match application UID (Service=10156 | Application=10167)
                D  Skipping duplicate provider com.enioka.scanner.sdk.zebra.EmdkZebraProvider : does not match application UID (Service=10156 | Application=10167)
                I  Provider discovery done
ViewRo...vity]  D  dispatchAttachedToWindow
                D  Relayout returned: oldFrame=[0,0][0,0] newFrame=[0,0][720,1280] result=0x27 surface={isValid=true -956407808} surfaceGenerationChanged=true
                D  mHardwareRenderer.initialize() mSurface={isValid=true -956407808} hwInitialized=true
OpenGLRenderer  I  Initialized EGL, version 1.4
                D  Swap behavior 1
ScannerService  D  Status changed: SERVICE_PROVIDER_SEARCH_OVER --- ScannerService a terminé la découverte de providers de scanner
                I  (re)starting scanner search!
LaserScanner    I  Starting scanner search
mali_winsys     D  EGLint new_window_surface(egl_winsys_display*, void*, EGLSurface, EGLConfig, egl_winsys_surface**, egl_color_buffer_format*, EGLBoolean) returns 0x3000,  [720x1280]-format:1
LaserScanner    D  Provider AthesiHHTProvider accepted
                D  Provider HoneywellOssIntegratedProvider accepted
                D  Provider AthesiE5LProvider accepted
                D  Provider ZebraDwProvider accepted
[... 100 lines cut ...]
ViewRo...vity]  D  mHardwareRenderer.destroy()#1
                D  Relayout returned: oldFrame=[0,0][720,1280] newFrame=[0,0][720,1280] result=0x5 surface={isValid=false 0} surfaceGenerationChanged=true
ViewRo...vity]  D  dispatchAttachedToWindow
                D  Relayout returned: oldFrame=[0,0][0,0] newFrame=[0,0][720,1280] result=0x7 surface={isValid=true -956407808} surfaceGenerationChanged=true
                D  mHardwareRenderer.initialize() mSurface={isValid=true -956407808} hwInitialized=true
mali_winsys     D  EGLint new_window_surface(egl_winsys_display*, void*, EGLSurface, EGLConfig, egl_winsys_surface**, egl_color_buffer_format*, EGLBoolean) returns 0x3000,  [720x1280]-format:1
Scanne...ivity  D  Service is connected to activity
ScannerService  D  Registering new client: com.geodis.mobicop.RegistrationScanActivity@6009cf0 100703472
                D  Notifying late clients that providers are already discovered
                D  Notifying late clients that scanners are already connected
Scanne...ivity  I  Activity can now use all received scanners (1)
ZebraD...vider  D  Sending intent to scanner to enable the trigger
ViewRo...vity]  D  MSG_RESIZED_REPORT: frame=Rect(0, 0 - 720, 1280) ci=Rect(0, 48 - 0, 0) vi=Rect(0, 48 - 0, 0) or=1
                D  MSG_WINDOW_FOCUS_CHANGED 1
                D  mHardwareRenderer.initializeIfNeeded()#2 mSurface={isValid=true -956407808}
InputM...nager  V  Starting input: tba=android.view.inputmethod.EditorInfo@5e2575 nm : com.geodis.mobicop ic=null
                I  [IMM] startInputInner - mService.startInputOrWindowGainedFocus
InputTransport  D  Input channel constructed: fd=68
                D  Input channel destroyed: fd=67
art             I  Do partial code cache collection, code=19KB, data=31KB
                I  After code cache collection, code=19KB, data=31KB
                I  Increasing code cache capacity to 128KB
ViewRo...vity]  D  mHardwareRenderer.destroy()#1
                D  Relayout returned: oldFrame=[0,0][720,1280] newFrame=[0,0][720,1280] result=0x1 surface={isValid=false 0} surfaceGenerationChanged=false
```

abelatar, 2023-10-05:
Hello @DaSpood , 
In version 2.3.9, the camera mode is activated on the TC27, even though it should operate in laser mode.

DaSpood, 2023-10-05:
Bonjour @abelatar , aurais-tu des traces d’exécution de l'application ? Le provider se base désormais sur l'existence du package datawedge (`com.symbol.datawedge`) pour détecter son existence et son activation.

abelatar, 2023-10-05:
Re @DaSpood 

Voici les traces : 
```
Provider BluebirdProvider was successfully instantiated
11:48:21.547                               D  Trying to instantiate provider com.enioka.scanner.sdk.honeywelloss.integrated.HoneywellOssIntegratedScannerProvider - not BT - 0
11:48:21.547                               D  Provider HoneywellOssIntegratedProvider was successfully instantiated
11:48:21.547                               D  Trying to instantiate provider com.enioka.scanner.sdk.proglove.ProgloveProvider - BT     - 0
11:48:21.547                               D  Provider ProgloveProvider was successfully instantiated
11:48:21.547                               D  Trying to instantiate provider com.enioka.scanner.bt.manager.SerialBtScannerProvider - BT     - 0
11:48:21.549                               D  Provider BtSppSdk was successfully instantiated
11:48:21.551 BtSppSdk                      D  Skipping duplicate provider com.enioka.scanner.sdk.generalscan.GsSppScannerProvider : does not match application UID (Service=10270 | Application=10271)
11:48:21.551                               D  Skipping duplicate provider com.enioka.scanner.sdk.zebraoss.ZebraOssSppScannerProvider : does not match application UID (Service=10270 | Application=10271)
11:48:21.551                               D  Skipping duplicate provider com.enioka.scanner.sdk.zebraoss.ZebraOssAttScannerProvider : does not match application UID (Service=10270 | Application=10271)
11:48:21.551                               D  Skipping duplicate provider com.enioka.scanner.sdk.honeywelloss.spp.HoneywellOssSppScannerProvider : does not match application UID (Service=10270 | Application=10271)
11:48:21.552                               I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.generalscan.GsSppScannerProvider
11:48:21.552                               I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.zebraoss.ZebraOssSppScannerProvider
11:48:21.552                               I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.zebraoss.ZebraOssAttScannerProvider
11:48:21.553                               I  	SPP SDK compatible provider found: com.enioka.scanner.sdk.honeywelloss.spp.HoneywellOssSppScannerProvider
11:48:21.553 LaserScanner                  I  Provider discovery done
11:48:21.568 ScannerService                D  Status changed: SERVICE_PROVIDER_SEARCH_OVER --- ScannerService finished discovering scanner providers
11:48:21.568                               I  (re)starting scanner search!
11:48:21.568 LaserScanner                  I  Starting scanner search
11:48:21.574                               D  Provider ProgloveProvider skipped because bluetooth option is disabled
11:48:21.574                               D  Provider ZebraDwProvider accepted
11:48:21.574                               D  Provider BluebirdProvider accepted
11:48:21.574                               D  Provider AthesiHHTProvider accepted
11:48:21.574                               D  Provider AthesiE5LProvider accepted
11:48:21.574                               D  Provider HoneywellOssIntegratedProvider accepted
11:48:21.574                               D  Provider BtSppSdk skipped because bluetooth option is disabled
11:48:21.574                               I  There are 5 providers which are going to be invoked for fresh laser scanners
11:48:21.574                               I  	ZebraDwProvider
11:48:21.574                               I  	HoneywellOssIntegratedProvider
11:48:21.574                               I  	BluebirdProvider
11:48:21.574                               I  	AthesiHHTProvider
11:48:21.574                               I  	AthesiE5LProvider
11:48:21.575                               I  Starting search on provider ZebraDwProvider
11:48:21.575 IntentProvider                D  Provider ZebraDwProvider: Checking appPackage `com.symbol.datawedge`
11:48:21.576 LaserScanner                  I  Starting search on provider BluebirdProvider
11:48:21.577 IntentProvider                D  Provider BluebirdProvider: Checking intent `kr.co.bluebird.android.bbapi.action.BARCODE_OPEN`
11:48:21.577                               D  Provider ZebraDwProvider: AppPackage `com.symbol.datawedge` not found
11:48:21.577 LaserScanner                  I  Starting search on provider HoneywellOssIntegratedProvider
11:48:21.578 IntentProvider                D  Provider ZebraDwProvider not compatible
11:48:21.578                               D  Provider HoneywellOssIntegratedProvider: Checking intent `com.honeywell.decode.DecodeService`
11:48:21.578 LaserScanner                  I  Scanner provider ZebraDwProvider reports it is not compatible with the device and will be disabled
11:48:21.579 IntentProvider                D  Provider HoneywellOssIntegratedProvider: Intent `com.honeywell.decode.DecodeService` not found
11:48:21.580 LaserScanner                  I  Starting search on provider AthesiE5LProvider
11:48:21.580 IntentProvider                D  Provider AthesiE5LProvider: Checking specific devices `[RD50TE]`
11:48:21.580                               D  Provider AthesiE5LProvider not compatible
11:48:21.580 LaserScanner                  I  Scanner provider AthesiE5LProvider reports it is not compatible with the device and will be disabled
11:48:21.580                               I  Starting search on provider AthesiHHTProvider
[... cut ...]
12:12:48.015                               I  Camera repeating capture request was set up 146149111
12:12:48.015                               D  Capture session has nothing to process 146149111
12:12:48.015                               D  Capture session is getting active 146149111
12:12:48.341                               D  Creating new buffer (MB) 1 count is 1
12:12:48.404                               D  FPS: 1.0031678E-4 - Pool size: 6. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
12:12:48.475 ScannerActivity               I  Scanner activity is 
```

marcanpilami, 2023-10-05:
This is weird. We have tested on a TC27 without issue. Are you sure you have not disabled to DW service between tests?

If it is enabled, can you list the installed packages on your TC27 device?
`./adb shell pm list packages -u symbol` should work

Also of interest to us: `./adb shell pm dump com.symbol.datawedge`

Finally: what is the target SDK version of your app?

marcanpilami, 2023-10-05:
Forget about it. It's a permission issue for 33+ apps. You could add an entry to the query section of you manifest, or simply upgrade to version 2.3.10, incoming in a few minutes.

commit 08413ec5 (2023-10-04): Fixed #124: ZebraDwProvider compatibility test

Use appPackageToTest instead of serviceToTest.
commit 4df25a86 (2023-10-04): Fixed #124: IntentProvider service test
commit 4f6de757 (2023-10-06): Fixed #124: IntentProvider service test
commit 6e4c4354 (2023-10-06): Fixed #124: ZebraDwProvider compatibility test

Use appPackageToTest instead of serviceToTest.
closed 2023-10-05 by marcanpilami