enioka-Haute-Couture/enioka_scan #135: A34 Problème avec le bouton de flash : allumage sans possibilité d'extinction

abelatar, 2023-10-16:
Lors de l'utilisation de notre application avec la librairie enioka scan 2.3.11 sur le modèle A34, nous rencontrons un problème avec le bouton de flash. Sur la plupart des téléphones, un double-clic sur le bouton permet d'activer puis de désactiver le flash. Cependant, sur le modèle A34, une fois que le flash est activé, il n'est pas possible de le désactiver en cliquant à nouveau sur le bouton.

Étapes pour reproduire :

Ouvrez l'écran de scan sur un A34 
Cliquez sur le bouton de flash pour l'activer
Tentez de cliquer à nouveau sur le bouton pour éteindre le flash (pas possible)

Pas de traces ou exceptions détectés sur le log : 

```
11:19:20.850 ViewRootImpl@...ingActivity]  I  ViewPostIme pointer 0
11:19:20.950                               I  ViewPostIme pointer 1
11:19:21.545 BARCODE                       D  FPS: 12.742358 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:22.602                               D  FPS: 29.352066 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:23.636                               D  FPS: 29.961906 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:24.675                               D  FPS: 29.858898 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:24.808 ViewRootImpl@...ingActivity]  I  ViewPostIme pointer 0
11:19:24.919                               I  ViewPostIme pointer 1
11:19:25.709 BARCODE                       D  FPS: 29.97865 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:26.749                               D  FPS: 29.812332 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:27.509 ViewRootImpl@...ingActivity]  I  ViewPostIme pointer 0
11:19:27.630                               I  ViewPostIme pointer 1
11:19:27.811 BARCODE                       D  FPS: 30.116 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:28.558 ViewRootImpl@...ingActivity]  I  ViewPostIme pointer 0
11:19:28.647                               I  ViewPostIme pointer 1
11:19:28.846 BARCODE                       D  FPS: 29.953146 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:29.827 ViewRootImpl@...ingActivity]  I  ViewPostIme pointer 0
11:19:29.881 BARCODE                       D  FPS: 29.93471 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:29.923 ViewRootImpl@...ingActivity]  I  ViewPostIme pointer 1
11:19:30.743                               I  ViewPostIme pointer 0
11:19:30.866                               I  ViewPostIme pointer 1
11:19:30.918 BARCODE                       D  FPS: 29.907806 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:31.627 ViewRootImpl@...ingActivity]  I  ViewPostIme pointer 0
11:19:31.741                               I  ViewPostIme pointer 1
11:19:31.955 BARCODE                       D  FPS: 29.88769 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:32.995                               D  FPS: 29.827797 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:34.030                               D  FPS: 29.934177 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:35.069                               D  FPS: 29.846897 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:36.104                               D  FPS: 29.938314 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:37.141                               D  FPS: 29.911953 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:38.174                               D  FPS: 29.998583 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:39.211                               D  FPS: 29.902613 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:40.246                               D  FPS: 29.934532 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:41.284                               D  FPS: 29.864243 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:42.318                               D  FPS: 29.986269 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:43.356                               D  FPS: 29.865425 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:44.391                               D  FPS: 29.958097 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:45.428                               D  FPS: 29.890467 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:46.465                               D  FPS: 29.894846 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:47.499                               D  FPS: 29.973185 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:48.537                               D  FPS: 29.88525 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:49.574                               D  FPS: 29.896933 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:50.607                               D  FPS: 29.991505 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:51.644                               D  FPS: 29.903774 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:52.681                               D  FPS: 29.899628 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:53.716                               D  FPS: 29.92817 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:54.514 ViewRootImpl@...ingActivity]  I  ViewPostIme pointer 0
11:19:54.541 BARCODE                       I  Left: 0 - Top : 534 - Right: 1080 - bottom: 688
11:19:54.541                               I  target was laid out
11:19:54.635 ViewRootImpl@...ingActivity]  I  ViewPostIme pointer 1
11:19:54.781 BARCODE                       D  FPS: 30.05453 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:55.715 ViewRootImpl@...ingActivity]  I  ViewPostIme pointer 0
11:19:55.808 BARCODE                       I  Left: 0 - Top : 535 - Right: 1080 - bottom: 689
11:19:55.808                               I  target was laid out
11:19:55.811 ViewRootImpl@...ingActivity]  I  ViewPostIme pointer 1
11:19:55.817 BARCODE                       D  FPS: 29.916126 - Pool size: 8. Current analysis queue depth: 0. Current res: 1440*1080. Statistics are: {}
11:19:55.826                               I  Left: 0 - Top : 536 - Right: 1080 - bottom: 690
11:19:55.826                               I  target was laid out
11:19:56.412 ViewRootImpl@...ingActivity]  I  ViewPostIme pointer 0
11:19:56.504                               I  ViewPostIme pointer 1
```

fmanoir, 2023-10-30:
Je pense que le problème 
> There are also some problems while turning On/Off flashlight. eg. for the devices not having FLASH_MODE_TORCH or even if it has, then flashlight does not turn ON etc.

donc il sera réglé avec l'utilisation : 

https://stackoverflow.com/questions/6939816/turn-on-off-camera-led-flash-light-in-samsung-galaxy-ace-2-2-1-galaxy-tab

fmanoir, 2023-11-03:
Bonjour  @DaSpood ,

La solution pour corriger cette issue, modifier : 

Classe CameraBarcodeScanViewV2 // Ligne 462 : 
De

`captureRequestBuilder.set(CaptureRequest.FLASH_MODE, null);`

à 

`captureRequestBuilder.set(CaptureRequest.FLASH_MODE,  CaptureRequest.FLASH_MODE_OFF);`

DaSpood, 2023-11-03:
> Bonjour @DaSpood ,
> 
> La solution pour corriger cette issue, modifier :
> 
> Classe CameraBarcodeScanViewV2 // Ligne 462 : De
> 
> `captureRequestBuilder.set(CaptureRequest.FLASH_MODE, null);`
> 
> à
> 
> `captureRequestBuilder.set(CaptureRequest.FLASH_MODE, CaptureRequest.FLASH_MODE_OFF);`

Ajouté a la branche `add-safeguards-for-camera` :+1:

DaSpood, 2023-11-17:
Normalement corrigé avec la 2.4.1, rouvrir une issue si nécessaire

closed 2023-11-17 by DaSpood