enioka-Haute-Couture/enioka_scan #137: Problème de fluidité de scan dû à un souci de focus sur Xcover 5

abelatar, 2023-10-16:
Hello @DaSpood 

Lors de l'utilisation de notre application sur le modèle Xcover 5, nous avons constaté un manque de fluidité lors du scan. Le souci semble être dû à un problème de focus sur la lib version 2.3.11. L'utilisateur est contraint de maintenir l'appareil à une distance très précise pour réussir à scanner correctement.

**Étapes pour reproduire :**
Ouvrez l'application sur un Xcover 5 et accédez à la fonction de scan.
Essayez de scanner un code-barres à différentes distances.

**Résultat attendu :**
La librairie devrait permettre un scan fluide et adaptatif, peu importe la distance relative entre l'appareil et le code-barres comme sur l'ancienne version de la lib avant de migrer vers Camera2 
```
14:32:22.533 ViewRootImpl@...ionActivity]  I  ViewPostIme pointer 0
14:32:22.627                               I  ViewPostIme pointer 1
14:32:22.673 ScannerActivity               I  Scanner activity is being paused 149184178
14:32:22.719                               D  Scanner activity is created 88027718
14:32:22.731 DecorView                     I  [INFO] isPopOver=false config=true
14:32:22.731                               I  updateCaptionType: isFloating=false isApplication=true hasWindowDecorCaption=false this=DecorView@cd4cca0[]
14:32:22.731                               D  setCaptionType = 0, this = DecorView@cd4cca0[]
14:32:22.736                               I  getCurrentDensityDpi: from real metrics. densityDpi=340 msg=resources_loaded
14:32:22.736                               I  setWindowBackground: isPopOver=false color=fffafafa d=android.graphics.drawable.ColorDrawable@77c441e
14:32:22.766 BARCODE                       D  Creating camera view layout
14:32:22.768                               I  Analyser (ZBar) is ready inside pool
14:32:22.768                               I  Analyser (ZBar) is ready inside pool
14:32:22.769                               I  Analyser (ZBar) is ready inside pool
14:32:22.769                               I  Analyser (ZBar) is ready inside pool
14:32:22.770                               I  Analyser (ZBar) is ready inside pool
14:32:22.772                               I  Analyser (ZBar) is ready inside pool
14:32:22.772                               I  Analyser (ZBar) is ready inside pool
14:32:22.773                               I  Analyser (ZBar) is ready inside pool
14:32:22.774                               I  Analyser pool initialized with 8 threads
14:32:22.777                               D  Camera2 specific initialization start
14:32:22.777                               D  Starting background thread handler 56539382
14:32:22.778 ScannerActivity               I  Using camera version: com.enioka.scanner.camera.CameraBarcodeScanViewV2
14:32:22.792 ClassificationActivity        I  first classification
14:32:22.799 ScannerActivity               D  Scanner activity is starting 88027718
14:32:22.821                               I  Scanner activity is resuming 88027718
14:32:22.822                               I  Resuming scanner activity in camera mode
14:32:22.822                               I  Giving up on laser, going to camera
14:32:22.829 MobiprepActivity              I  Wifi is enabled
14:32:22.838 MSHandlerLifeCycle            I  check: return. pkg=com.geodis.mobiprep parent=null callers=com.android.internal.policy.DecorView.setVisibility:4412 android.app.ActivityThread.handleResumeActivity:5476 android.app.servertransaction.ResumeActivityItem.execute:54 android.app.servertransaction.ActivityTransactionItem.execute:45 android.app.servertransaction.TransactionExecutor.executeLifecycleState:176 
14:32:22.838                               I  removeMultiSplitHandler: no exist. decor=DecorView@cd4cca0[]
14:32:22.845 NativeCustomFrequencyManager  D  [NativeCFMS] BpCustomFrequencyManager::BpCustomFrequencyManager()
14:32:22.855 InsetsController              D  onStateChanged: InsetsState: {mDisplayFrame=Rect(0, 0 - 720, 1480), mDisplayCutout=DisplayCutout{insets=Rect(0, 0 - 0, 0) waterfall=Insets{left=0, top=0, right=0, bottom=0} boundingRect={Bounds=[Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0)]} cutoutPathParserInfo={CutoutPathParserInfo{displayWidth=0 displayHeight=0 physicalDisplayWidth=0 physicalDisplayHeight=0 density={0.0} cutoutSpec={} rotation={0} scale={0.0} physicalPixelDisplaySizeRatio={0.0}}}}, mRoundedCorners=RoundedCorners{[RoundedCorner{position=TopLeft, radius=68, center=Point(68, 68)}, RoundedCorner{position=TopRight, radius=68, center=Point(652, 68)}, RoundedCorner{position=BottomRight, radius=68, center=Point(652, 1412)}, RoundedCorner{position=BottomLeft, radius=68, center=Point(68, 1412)}]}  mRoundedCornerFrame=Rect(0, 0 - 720, 1480), mPrivacyIndicatorBounds=PrivacyIndicatorBounds {static bounds=Rect(626, 0 - 720, 51) rotation=0}, mSources= { InsetsSource: {mType=ITYPE_STATUS_BAR, mFrame=[0,0][720,51], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_NAVIGATION_BAR, mFrame=[0,1378][720,1480], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_LEFT_GESTURES, mFrame=[0,0][0,1480], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_RIGHT_GESTURES, mFrame=[720,0][720,1480], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_TOP_MANDATORY_GESTURES, mFrame=[0,0][720,51], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_BOTTOM_MANDATORY_GESTURES, mFrame=[0,1378][720,1480], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_TOP_TAPPABLE_ELEMENT, mFrame=[0,0][720,51], mVisible=true, mInsetsRoundedCornerFrame=false}, InsetsSource: {mType=ITYPE_BOTTOM_TAPPABLE_ELEMENT, mFrame=[0,1378][720,1480], mVisible=true, mInsetsRoundedCornerFrame=false} } host=com.geodis.mobiprep/com.geodis.mobiprep.ClassificationActivity from=android.view.ViewRootImpl.setView:1732
14:32:22.859 ViewRootImpl@...ionActivity]  I  setView = com.android.internal.policy.DecorView@cd4cca0 TM=true
14:32:22.861 MSHandlerLifeCycle            I  removeMultiSplitHandler: no exist. decor=DecorView@cd4cca0[ClassificationActivity]
14:32:22.869 SurfaceView@2667191           I  onWindowVisibilityChanged(0) true com.enioka.scanner.camera.CameraPreviewSurfaceView{2667191 V.E...... ......I. 0,0-0,0} of ViewRootImpl@5cff0ce[ClassificationActivity]
14:32:22.869                               D  updateSurface: has no frame
14:32:22.883 ViewRootImpl@...ionActivity]  I  performTraversals params={(0,0)(fillxfill) sim={adjust=pan forwardNavigation} ty=BASE_APPLICATION wanim=0x1030309
                                                fl=81810100
                                                pfl=12020040
                                                bhv=DEFAULT
                                                fitSides= naviIconColor=0
                                                sfl=100000}
14:32:22.887                               I  performTraversals mFirst=true windowShouldResize=true viewVisibilityChanged=false mForceNextWindowRelayout=false params={(0,0)(fillxfill) sim={adjust=pan forwardNavigation} ty=BASE_APPLICATION fmt=TRANSLUCENT wanim=0x1030309
                                                fl=81810100
                                                pfl=12020040
                                                bhv=DEFAULT
                                                fitSides= naviIconColor=0
                                                sfl=100000}
14:32:22.909 InsetsController              D  onStateChanged: InsetsState: {mDisplayFrame=Rect(0, 0 - 720, 1480), mDisplayCutout=DisplayCutout{insets=Rect(0, 0 - 0, 0) waterfall=Insets{left=0, top=0, right=0, bottom=0} boundingRect={Bounds=[Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0)]} cutoutPathParserInfo={CutoutPathParserInfo{displayWidth=0 displayHeight=0 physicalDisplayWidth=0 physicalDisplayHeight=0 density={0.0} cutoutSpec={} rotation={0} scale={0.0} physicalPixelDisplaySizeRatio={0.0}}}}, mRoundedCorners=RoundedCorners{[RoundedCorner{position=TopLeft, radius=68, center=Point(68, 68)}, RoundedCorner{position=TopRight, radius=68, center=Point(652, 68)}, RoundedCorner{position=BottomRight, radius=68, center=Point(652, 1412)}, RoundedCorner{position=BottomLeft, radius=68, center=Point(68, 1412)}]}  mRoundedCornerFrame=Rect(0, 0 - 720, 1480), mPrivacyIndicatorBounds=PrivacyIndicatorBounds {static bounds=Rect(626, 0 - 720, 51) rotation=0}, mSources= {  [... cut ...]
14:34:44.675                               D  FPS: 16.727978 - Pool size: 8. Current analysis queue depth: 0. Current res: 1920*1080. Statistics are: {Point(1920, 1080)=28}
14:34:45.762                               D  FPS: 16.55155 - Pool size: 8. Current analysis queue depth: 0. Current res: 1920*1080. Statistics are: {Point(1920, 1080)=28}
14:34:46.827                               D  FPS: 16.905317 - Pool size: 8. Current analysis queue depth: 0. Current res: 1920*1080. Statistics are: {Point(1920, 1080)=28}

```

DaSpood, 2023-10-26:
Pas pu reproduire sur nos devices, après discussion et update a la version 2.4.0 il semble que forcer la Camera1 sur ce device ait résolu le problème. Je vous laisse fermer l'issue si c'est bon

fmanoir, 2023-10-26:
Merci pour votre retour ; 

Avec le Xcover 5 = il faut se mettre à 15-17 cm pour scaner un code contre 10-11 cm sur le Xcover 4s ? il n'est pas possible d’ d’unifier le comportement de scan avec Camera2

closed 2023-11-02 by DaSpood