enioka-Haute-Couture/enioka_scan #139: Plantage non répétitif

fmanoir, 2023-10-26:
Bonjour @DaSpood, @marcanpilami ,

Lorsque nous exécutons par foi un scan en utilisant la librairie au sein de notre application, notamment sur les appareils Xcover 4S et 5, nous rencontrons un problème d'instabilité qui conduit à un crash de l'application. Voici les traces que j'ai pu récupérer suite à l'arrêt inattendu de l'application :

 [image]
 
**Proposition pour une solution**
Je pense qu'il faut ajouter un test if captureRequestBuilder != null entre ce bloc : 

`captureRequestBuilder.removeTarget(this.camPreviewSurfaceView.getHolder().getSurface());
captureRequestBuilder = null;
`

fmanoir, 2023-10-30:
Même souci que #134 avec A34 +  Xcover 4S et Xcover 5

DaSpood, 2023-11-17:
Normalement corrigé avec la 2.4.1, rouvrir une issue si nécessaire

closed 2023-11-17 by DaSpood