enioka-Haute-Couture/enioka_scan #141: FLASH s'allume + Plantage  pour un changment de rotation de portrait vers landscape

fmanoir, 2023-10-30:
Lorsqu'il y a un changement de rotation de portrait vers landscape, le FLASH s'allume.

Voici les traces que j'ai pu récupérer suite à l'arrêt inattendu de l'application :

[image]

**Je pense que l'instance caméra (cam) devient nulle donc il ne peut pas récupérer les paramètres**

fmanoir, 2023-10-30:
C'est même crach liée au #140

fmanoir, 2023-10-30:
Pour le même scénario il y a aussi par fois, un autre crach sur A34 : 

[image]

DaSpood, 2023-11-17:
Normalement corrigé avec la 2.4.1, rouvrir une issue si nécessaire

closed 2023-11-17 by DaSpood