enioka-Haute-Couture/enioka_scan #148: Amélioration : Réinitialiser l'emplacement de la target

fmanoir, 2023-12-14:
Bonjour, 

Est-ce qu'il serait envisageable d'apporter une petite amélioration lorsque allowTargetDragest activé ? qui nous permet de réinitialiser l'emplacement de la target (Rectangle rouge).

**Problème :** 

Lorsque je modifie l'emplacement de "target" avec un view au-dessus de camPreviewSurfaceView avec hauteur changeable, le target se cache derrière le view.

[image]

DaSpood, 2024-01-10:
L'ajout d'une méthode publique de la CameraBarcodeScanView (donc accessible via `findViewById(cameraViewId)` depuis l’activité) "resetTargetPosition()" qui réinitialiserait la position du rectangle (et aussi les preferences sauvegardées) au centre de l'image ferait-elle l'affaire ?

fmanoir, 2024-01-18:
Merci fonctionne parfaitement

closed 2024-01-18 by fmanoir