enioka-Haute-Couture/enioka_scan #150: Problème de distorsion

fmanoir, 2024-01-11:
Bonjour,

Bonjour @DaSpood, @marcanpilami 

Je trouve un problème de distorsion sur Xcover 5 / Xcover 6, existe même sur votre démo

L'image s'affiche correctement après la première ouverture, mais si nous quittons l'interface et la réouvrons, il y a le problème de distorsion (Xcover 6) 

Pour Xcover 5 c'est inversement  distorsion -> quittons l'interface ou background -> réouvrons : Ok 

[image]
[image]

DaSpood, 2024-01-12:
Bonjour @fmanoir 

Retours sur les problèmes de distortion: ces problèmes viennent du fait que la preview de la camera essaie de remplir l’écran peu importe sa resolution, ce qui provoque soit un étirement soit un aplatissement de l'image si sa resolution utilise un aspect ratio qui ne correspond pas exactement a la place disponible. Cette resolution peut aussi changer au fil du temps si les performances sont trop basses et que l'option `useAdaptiveResolution` est active, ce qui cause cette impression que l'image se degrade d'un essai a l'autre (la resolution est en fait modifiée, et le nouvel aspect ratio est plus ou moins proche de celui de la zone disponible compare a la precedente resolution).

Pour désactiver l’étirement de la photo, il est possible de changer dans le fichier xml du layout camera la valeur `app:previewRatioMode` a `fitToPicture` au lieu de `fillAvailableSpace` qui est sa valeur par défaut. Il est également possible de diminuer la valeur de `maxDistortionRatio` pour réduire la tolerance de deformation, mais le risque est qu'aucune des resolutions disponible ne soit suffisamment proche, auquel cas la bibliothèque les essaie une par une.

[image]

Avec ce paramètre, au lieu de chercher a remplir l'espace en s’étirant, l'image conservera son format, et aura des barres noires soit verticales soit horizontales si il reste de l'espace vide (comme lorsque l'on regarde un film au format 21:9 sur un écran de télé en 16:9).

J'ai effectué plusieurs tentatives de changements de resolution pour tenter de faire correspondre la resolution avec la place disponible mais la camera ne semble pas accepter de resolution ne correspondant pas a un format photo classique, dont l'aspect ratio correspond rarement a celui de l'application. Il est probablement possible de "crop" dynamiquement la preview mais je n'ai pas trouve de manière simple de le faire pour le moment.

fmanoir, 2024-01-16:
Bonjour, 

Je vous remercie pour votre réponse, j'ai essayé d'utiliser votre solution mais malheureusement, elle a eu le même comportement qu'auparavant.

```
  <com.enioka.scanner.camera.CameraBarcodeScanView
            android:id="@+id/camera_scan_view"
            android:layout_width="match_parent"
            android:layout_height="wrap_content"
            android:layout_below="@id/bar"
            android:layout_alignParentBottom="true"
            app:forceCameraApiVersion="Auto"
            app:maxResolutionY="1080"
            app:minResolutionY="720"
            app:readerMode="Auto"
            app:previewRatioMode="fitToPicture"
            app:storePreferredResolution="true"
            app:targetColorActive="@color/colorRed"
            app:targetColorPaused="@color/defaultItemColor"
            app:targetIsFixed="false"
            app:targetStrokeWidth="5"
            app:useAdaptiveResolution="true">
```

DaSpood, 2024-01-16:
Je prendrai un peu de temps demain apres-midi pour essayer de reproduire, meme si je vois d'apres #151 que CameraV1 ne semble pas avoir ce problème.

fmanoir, 2024-01-16:
Oui avec Camera V1 C'est Ok ; le souci avec Camera V2 uniquement.

DaSpood, 2024-01-17:
Un fix est disponible dans la version 2.4.2, qui sera disponible sur maven central sous peu. Il devrait éviter la distortion en mode `fitToPicture` (mais n'affecte pas le mode `fillAvailableSpace`). J'ai testé ce fix sur le XCover 4 qui avait bien un problème d’étirement meme en `fitToPicture` avant.

fmanoir, 2024-01-18:
Malheureusement, il me semble que même souci avec 2.4.2  avec  

- previewRatioMode de fillAvailableSpace  vers fitToPicture 
- forçage de V1 vers Auto 

[image]

DaSpood, 2024-01-19:
Malheureusement j'ai peur de ne pas pouvoir aller plus loin sans le device exact qui a ce problème, il s'agit vraisemblablement d'un bug lie a la resolution de l’écran. J'avais pu le reproduire sur mes appareils mais le problème est corrigé de mon coté, et visiblement pas du votre.

Je laisse l'issue ouverte pour le moment mais la solution temporaire de forcer CameraV1 risque de devoir durer un moment, désolé

fmanoir, 2024-01-19:
Oui, nous adaptons la solution temporaire de forcer CameraV1 mais jusqu'à quand :/   

Sinon Je reproduis le problème sur tous les smartphones , même mon propre Samsung, il suffit de faire une rotation en mode paysage -> portrait et tu trouveras le problème. 

La seule différence que je constate est que sur certains smartphones, le rendu est bon mais il y a un problème de distorsion lors de la rotation comme (Xcover6 , Samsung A54 , S21 ...) et d'autres le rendu directement distordu mais après un passage en arrière-plan, il est devenu bon

DaSpood, 2024-01-23:
@yurivital

DaSpood, 2024-04-11:
Update:

`currentPreviewResolution` semble ne pas toujours etre set des la creation de la surface, donc la methode `CameraPreviewSurfaceView.onMeasure()` n'applique pas les bons parametres a la surface. Rafraichir l'ecran a nouveau (via verrouillage ou rotation) semble corriger le probleme, mais plus de tests doivent etre faits pour trouver un vrai fix dans le code.

Devra attendre la 3.0.

DaSpood, 2024-10-23:
Bonjour,

Je me permet de fermer cette issue qui est sensee etre resolue depuis la release 3.0. En cas de probleme sur le meme sujet n'hesitez pas a creer une nouvelle issue et a referencer celle ci.

closed 2024-10-23 by DaSpood