enioka-Haute-Couture/enioka_scan #158: Certains CAB ne sont pas compatibles avec Enioka Scan.

fmanoir, 2024-02-09:
Bonjour @DaSpood, @marcanpilami

Certains clients nous ont signalé que Enioka Scan ne fonctionne pas avec certains CAB spécifiques. Pouvez-vous me donner une idée de la raison pour laquelle ce problème se produit?

**Exemple :** 

[image]

DaSpood, 2024-02-19:
Bonjour @fmanoir 

Il est possible qu'il s'agisse d'un problème de symbologie. Actuellement, enioka Scan ne supporte officiellement que les symbologies Code128, Code39, Dis2of5, Int2of5, Ean13, QrCode et Aztec2D. Bien qu'a premiere vue ceux de ton image utilisent Code39.

Il est aussi possible que ce soit du a un problème de lisibilité si le code est trop dégradé. Ou si le code est trop compact, la caméra peut avoir du mal a l'identifier.

Quel appareil est utilisé pour ce cas d'usage ? Et la symbologie est-elle bien supportée ?

fmanoir, 2024-02-19:
Xcover4s / XCover5

Test avec EniokaScan v1.2.0 => La lecture du CAB fonctionne

Test avec EniokScan v2.4.3 => la lecture du CAB ne fonctionne pas.

fmanoir, 2024-02-23:
Je trouve le même souci avec Code39 dans votre demo

closed 2024-04-08 by fmanoir