enioka-Haute-Couture/enioka_scan #159: Enioka Scan & Jetpack Compose

fmanoir, 2024-02-16:
Bonjour, 

Je remarque que la librairie n'est pas compatible avec Jetpack Compose, est-ce qu'il y a une idée d'ajouter un support?

DaSpood, 2024-02-19:
Bonjour @fmanoir 

Pas de plan concret pour le moment, mais le support pour Jetpack Compose sera probablement ajouté lors de la migration vers Android X. Ce n'est cependant pas un sujet très prioritaire pour le moment.

DaSpood, 2024-05-13:
Update sur cette issue: la bbliotheque sera migree vers AndroidX pour la release 3.0.0, qui sera normalement disponible cet ete.

DaSpood, 2024-10-23:
Bonjour,

Je me permet de fermer cette issue qui est sensee etre resolue depuis la release 3.0. En cas de probleme sur le meme sujet n'hesitez pas a creer une nouvelle issue et a referencer celle ci.

closed 2024-10-23 by DaSpood