enioka-Haute-Couture/enioka_scan #174: Pop-Up Saisie manuelle

fmanoir, 2024-04-18:
Bonjour @DaSpood

Est-il possible d'ajouter un paramètre dans le constructeur pour changer la couleur du bouton et l'input?

Le style de nos applications est bleu. Il ne nous reste que cette Pop-Up Saisie manuelle  en Verte :

[image]

fmanoir, 2024-05-14:
Bonjour @DaSpood 

Oui c'est Ok pour barre horizontale  avec colorAccent  
mais c'est KO pour le bouton avec (@+id/scanner_manual_bt_ok) , si possible de le modifier pour qu'il avoir colorAccent  aussi  par défaut 

Merci.

DaSpood, 2024-10-23:
Bonjour,

Je me permet de fermer cette issue qui est sensee etre resolue depuis la release 3.0. En cas de probleme sur le meme sujet n'hesitez pas a creer une nouvelle issue et a referencer celle ci.

closed 2024-10-23 by DaSpood