enioka-Haute-Couture/enioka_scan #187: Possible to add / access the position of target rectangle in CameraBarcodeScanView

KFARINAALI, 2024-05-28:
Bonjour,

**Problem** : 
According to the UI below , we want to show the camera on the top part only but the scaling of the camera does not allow proper scanning. So to fix this we created an overlay over the camera , with camera full screen; In this case the red rectangle -> Target rectangle is by default in the center. In our case it is hidden behind the bottom overlay.

[image]

It is possible to access the target rectangle or change the default position of the rectangle.  It will be helpful if we can set default position . 

Merci

DaSpood, 2024-05-29:
I added two issues that should both provide a different way to fix the issue:

- #188 which would let you move the target
- #189 which would let you set the camera view to half the screen size with limited distortion and no black bars, at the cost of a cropped image

The latter will probably take a bit longer to implement and will likely be released after version 3.0, but the first one should be doable in time for next release.

DaSpood, 2024-10-23:
Bonjour,

Je me permet de fermer cette issue qui est sensee etre resolue depuis la release 3.0. En cas de probleme sur le meme sujet n'hesitez pas a creer une nouvelle issue et a referencer celle ci.

closed 2024-10-23 by DaSpood