enioka-Haute-Couture/enioka_scan #230: Error receiving broadcast intent : Datawedge

KFARINAALI, 2025-04-18:
Exception issue : 

[image]

While observing logs for our application MobiPrep we often see this Runtime exception caused by broadcast intent.
There is some problem of communication between datawedge and the application. 

Please take a look 

Thank you

DaSpood, 2025-06-30:
Hello @KFARINAALI 
This log was very helpful, I have a better idea of where to look
Could you check if, for devices where the error occurs, this log is also present:
```
Could not query config - empty result
```
It is a Warning level log with the `ZebraDwScanner` tag.

I should be able to push a fix to prevent the exceptions soon.

KFARINAALI, 2025-07-04:
Hi @DaSpood 
So far I could not find this warning log.

closed 2025-10-20 by DaSpood